AFM-SPM/TopoStats #362: Clear completion message

derollins, 2022-11-15:
**Is your feature request related to a problem? Please describe.**
Starting to use Topostats it is hard to know if the programme has finished running successfully or it has just crashed since the final line currently looks the same as all the rest but just says it had written a config file. 

**Describe the solution you'd like**
It would be great to have a clear (friendly!) message that lets you know that the programme has finished running without crashing. This could also highlight significant errors if there were any.

ns-rse, 2022-11-15:
This is a good idea, thanks for the suggestion @derollins 

I would propose that because the design is to process batches of scans and (hopefully!) if one (or more) images fails to process others will complete then a summary of the number of scans that have successfully processed and the number that have failed are reported at the end.

The file `all_statistics.csv` will contain grain and tracing statistics on those that have successfully processed but a list of those that failed would no doubt be useful. I think more generally there is scope to improve error messages when failures occur, but with limited testing this will come naturally as it sees wider usage, errors crop up and we work out what has caused them (and hopefully how to solve them!).

SylviaWhittle, 2023-01-10:
Does the completion message `~~~~~~~~~~~~~~~~~~~~ COMPLETE ~~~~~~~~~~~~~~~~~~~~` that was added in your PR #416 satisfy this issue and therefore should it be closed, @ns-rse?

Jean-Du, 2023-01-11:
> Does the completion message `~~~~~~~~~~~~~~~~~~~~ COMPLETE ~~~~~~~~~~~~~~~~~~~~` that was added in your PR #416 satisfy this issue and therefore should it be closed, @ns-rse, or do we want error listing and bad file warnings added too?

We need the experimentalists' input on this. @derollins Does this new completion message suit experimentalists' needs?

derollins, 2023-01-11:
I think this is good but 'complete' could be misleading if some of the files have failed maybe

ns-rse, 2023-01-11:
> It would be great to have a clear (friendly!) message that lets you know that the programme has finished running without crashing. This could also highlight significant errors if there were any.

The change addressed the first part as giving a clear message that the run completes without crashing.

For the second part until we have a good and broad understanding of the many different ways in which TopoStats might fail I don't think we can give too much detailed information. This is an on-going process and only really possible since TopoStats has started to see wider use in the last few weeks (developing on a single image is somewhat limiting).

A crude summary of how many files detected (which has always appeared at the top of the output and generated log-file), how many processed to completion and how many failed would be possible in the mean time though.

PR #434 (2023-01-13): Corrects image paths, lints code, adds command line options and more
Closes #362 
Closes #391 
Closes #428 
Closes #430 
Closes #433 

Pull request and code changes grew after initially making #429 (see for useful comments/feedback) but in summary...

## Double periods

`LoadScans()` class gets the filename using `Path.stem` method. However `processed_scan()` in `run_topostats.py` also took `Path.stem` so if filenames had two (or more) periods in them the filename ended up truncated and the last two sections after periods were excised (e.g. `mini.circle.spm` > `mini`). This is resolved by usin [... cut]
closed 2023-01-13 by ns-rse