AFM-SPM/TopoStats #366: Failure of DNA tracing results in no grainstats produced

MaxGamill-Sheffield, 2022-11-17:
**Describe the bug**
Due to the broad try/except statement over both grainstats and dnatracing, if molecules are found and something within dnatracing breaks, all statistics are lost as they are replaced with the empty dataframe. This is particularly an issue for #365 as @Jean-Du has mentioned that the changes in this PR don't work for all molecules and thus produces an error.

**To Reproduce**
Add a line in dnatracing which should throw an error (to simulate dnatracing breaking) and run topostats.
The resulting statistics file will now be empty.

**Expected behaviour**
It should at least contain the statistics already obtained from grainstats.py

**Output**
Empty dataframe for the image & thus an empty `all_stats` csv file.

**(Not required if using Docker) Your computer configuration (please complete the following information):**

**Additional context**
Wether this should be added on a per-function level, rather than the whole dnatracing statistics level should be discussed. 
This would then enable most of the dnatracing outputs to function correctly and be produced in the csv while only loosing those which fail.
Alternatively, the functions should work for all DNA molecules, however, this may be a good way to avoid future problems for edge cases.

ns-rse, 2023-12-19:
Just shy of a year late, but I think this might have been at least partially solved by @SylviaWhittle in #613 

Looking at the code the [`topostats.processing.run_dnatracing()`](https://github.com/AFM-SPM/TopoStats/blob/1b6e83aa2a8d73d6dd2a9491faa7d861b2064e0b/topostats/processing.py#L349) function still has a broad `try: ... except: ...`.

If an exception is encountered then the [`results = results_df`](https://github.com/AFM-SPM/TopoStats/blob/1b6e83aa2a8d73d6dd2a9491faa7d861b2064e0b/topostats/processing.py#L476) which will be just the `grainstats` dataframe that is passed into the function and this  is returned.

Thus whilst we haven't got fine-grained `try: ... except:` throughout the DNA tracing stages if it does fail the grain statistics should be returned.

If anyone has an image that they know fails tracing could they share a link and I will endeavour to test whether this is the case (more than welcome to try if you've time to do so).

Thanks in advance,

Neil

MaxGamill-Sheffield, 2024-04-03:
Closing as fixed (in possibly #613). 

Confirmed by adding `assert False` statements at multiple stages within dnatracing.py

closed 2024-04-03 by MaxGamill-Sheffield