AFM-SPM/TopoStats #367: Colour maps slightly wrong

SylviaWhittle, 2022-11-17:
**Describe the bug**
@derollins pointed out to me that the colour maps for new Topo are still slightly off. See attached images for comparison (left - old topo, right - new topo). This hinders work on projects that utilised the old TopoStats. Eddie also believes that the old colour scheme is better, so if the current scheme could be adjusted to match the previous one, that would be great. 

[image]

**(Not required if using Docker) Your computer configuration (please complete the following information):**
- All computer configurations
- All Python versions

MaxGamill-Sheffield, 2022-12-15:
I think I know how to fix the nanoscope colour gradient thing, or at least find the correct gradient values. When someone has time, please could someone get these via:
1. Open Gwyddion
2. Edit > Colour gradients
3. Click the Nanoscope option
4. Click “create new item based on selected one”
5. Click the new one
6. Click “edit selected item”

Now for the tricky bit:
1. Let me know if the arrows have the same spacing (this will be trickier if not)
2. Click each of the triangles and let me know the RGB values

MaxGamill-Sheffield, 2023-02-17:
Closed as resolved in #409

closed 2023-02-17 by MaxGamill-Sheffield

ns-rse, 2023-02-17:
Sorry for not linking these two, had dropped out of visibility having been merged. One more closed issue closer to a new release. :tada: