AFM-SPM/TopoStats #381: Plane tilt removal crash

SylviaWhittle, 2022-11-23:
**Describe the bug**
In rare cases, some images fail to process due to the line of best fit in the plane flattening steps not being able to return a value. 

**To Reproduce**

As the data is under NDA, I cannot give instructions on how to reproduce this. Apologies.

I believe that the error stems from a large scar reaching almost completely across from the left hand side to the right hand side of the image, however I cannot be sure as I have been unable to re-create the crash since I do not have the config file used. I will keep investigating though. 

The error itself stems from the calculation of the slant in the `y` (vertical) direction, from `[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] x-polyfit 1st order: [ 0.03401922 -8.87987985]
/Users/libbyholmes/opt/anaconda3/envs/topostats/lib/python3.10/site-packages/numpy/lib/polynomial.py:667: RuntimeWarning: invalid value encountered in divide`

**Output**

```
[Thu, 17 Nov 2022 15:58:33] [INFO  ] [topostats] Configuration is valid.
[Thu, 17 Nov 2022 15:58:33] [INFO  ] [topostats] Plotting configuration is valid.
[Thu, 17 Nov 2022 15:58:33] [INFO  ] [topostats] Configuration file loaded from   : libbys_config.yaml
[Thu, 17 Nov 2022 15:58:33] [INFO  ] [topostats] Scanning for images in       : Data/20221108_catenanes
[Thu, 17 Nov 2022 15:58:33] [INFO  ] [topostats] Output directory          : Data/20221108_catenanes
[Thu, 17 Nov 2022 15:58:33] [INFO  ] [topostats] Looking for images with extension  : .spm
[Thu, 17 Nov 2022 15:58:33] [INFO  ] [topostats] Images with extension .spm in Data/20221108_catenanes : 26
[Thu, 17 Nov 2022 15:58:33] [INFO  ] [topostats] Thresholding method (Filtering)   : std_dev
[Thu, 17 Nov 2022 15:58:33] [INFO  ] [topostats] Thresholding method (Grains)    : std_dev
Processing images from Data/20221108_catenanes, results are under Data/20221108_[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] Processing : Data/20221108_catenanes/20220811_nickedcatenane_6ng_nicl2_mgcl2.0_00013.spm
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] Extracting image from .spm
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] Processing : Data/20221108_catenanes/20220811_nickedcatenane_6ng_nicl2_mgcl2.0_00038.spm
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] Loading image from : Data/20221108_catenanes/20220811_nickedcatenane_6ng_nicl2_mgcl2.0_00013.spm
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] Extracting image from .spm
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] Loading image from : Data/20221108_catenanes/20220811_nickedcatenane_6ng_nicl2_mgcl2.0_00038.spm
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] [20220811_nickedcatenane_6ng_nicl2_mgcl2.0_00038] : Loaded image from : Data/20221108_catenanes/20220811_nickedcatenane_6ng_nicl2_mgcl2.0_00038.spm
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] [20220811_nickedcatenane_6ng_nicl2_mgcl2.0_00013] : Loaded image from : Data/20221108_catenanes/20220811_nickedcatenane_6ng_nicl2_mgcl2.0_00013.spm
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] Processing : Data/20221108_catenanes/20220811_nickedcatenane_6ng_nicl2_mgcl2.0_00039.spm
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] Extracting image from .spm
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] Loading image from : Data/20221108_catenanes/20220811_nickedcatenane_6ng_nicl2_mgcl2.0_00039.spm
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] Processing : Data/20221108_catenanes/20220811_nickedcatenane_6ng_nicl2_mgcl2.0_00029.spm
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] Extracting image from .spm
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] Loading image from : Data/20221108_catenanes/20220811_nickedcatenane_6ng_nicl2_mgcl2.0_00029.spm
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] [20220811_nickedcatenane_6ng_nicl2_mgcl2.0_00039] : Loaded image from : Data/20221108_catenanes/20220811_nickedcatenane_6ng_nicl2_mgcl2.0_00039.spm
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] [20220811_nickedcatenane_6ng_nicl2_mgcl2.0_00039] : Extracted channel Height
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] [20220811_nickedcatenane_6ng_nicl2_mgcl2.0_00039] : Pixel to nm scaling : 1.953125
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] median flattening without mask
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] [20220811_nickedcatenane_6ng_nicl2_mgcl2.0_00029] : Loaded image from : Data/20221108_catenanes/20220811_nickedcatenane_6ng_nicl2_mgcl2.0_00029.spm
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] plane removal without mask
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] x-polyfit 1st order: [ 0.03401922 -8.87987985]
/Users/libbyholmes/opt/anaconda3/envs/topostats/lib/python3.10/site-packages/numpy/lib/polynomial.py:667: RuntimeWarning: invalid value encountered in divide
 lhs /= scale
 ** On entry to DLASCLS parameter number 4 had an illegal value
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] Processing : Data/20221108_catenanes/20220811_nickedcatenane_6ng_nicl2_mgcl2.0_00015.spm
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] Extracting image from .spm
[Thu, 17 Nov 2022 15:58:37] [INFO  ] [topostats] Loading image from : Data/20221108_catenanes/20220811_nickedcatenane_6ng_nicl2_mgcl2.0_00015.spm
Processing images from Data/20221108_catenanes, results are under Data/20221108_
multiprocessing.pool.RemoteTraceback:
"""
Traceback (most recent call last):
 File "/Users/libbyholmes/opt/anaconda3/envs/topostats/lib/python3.10/multiprocessing/pool.py", line 125, in worker
  result = (True, func(*args, **kwds))
 File "/Users/libbyholmes/opt/anaconda3/envs/topostats/lib/python3.10/site-packages/topostats/run_topostats.py", line 200, in process_scan
  filtered_image.filter_image()
 File "/Users/libbyholmes/opt/anaconda3/envs/topostats/lib/python3.10/site-packages/topostats/filters.py", line 360, in filter_image
  self.images["initial_tilt_removal"] = self.remove_tilt(self.images["initial_align"], mask=None)
 File "/Users/libbyholmes/opt/anaconda3/envs/topostats/lib/python3.10/site-packages/topostats/filters.py", line 175, in remove_tilt
  py = np.polyfit(range(0, len(medians_y)), medians_y, 1)
 File "<__array_function__ internals>", line 180, in polyfit
 File "/Users/libbyholmes/opt/anaconda3/envs/topostats/lib/python3.10/site-packages/numpy/lib/polynomial.py", line 668, in polyfit
  c, resids, rank, s = lstsq(lhs, rhs, rcond)
 File "<__array_function__ internals>", line 180, in lstsq
 File "/Users/libbyholmes/opt/anaconda3/envs/topostats/lib/python3.10/site-packages/numpy/linalg/linalg.py", line 2300, in lstsq
  x, resids, rank, s = gufunc(a, b, rcond, signature=signature, extobj=extobj)
 File "/Users/libbyholmes/opt/anaconda3/envs/topostats/lib/python3.10/site-packages/numpy/linalg/linalg.py", line 101, in _raise_linalgerror_lstsq
  raise LinAlgError("SVD did not converge in Linear Least Squares")
numpy.linalg.LinAlgError: SVD did not converge in Linear Least Squares
"""
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
 File "/Users/libbyholmes/opt/anaconda3/envs/topostats/bin/run_topostats", line 8, in <module>
  sys.exit(main())
 File "/Users/libbyholmes/opt/anaconda3/envs/topostats/lib/python3.10/site-packages/topostats/run_topostats.py", line 435, in main
  for img, result in pool.imap_unordered(processing_function, img_files):
 File "/Users/libbyholmes/opt/anaconda3/envs/topostats/lib/python3.10/multiprocessing/pool.py", line 870, in next
  raise value
```

**(Not required if using Docker) Your computer configuration (please complete the following information):**
Reporting on behalf of Libby, so I don't know some of the details as they were not recorded at the time.
 - OS: MacOS
 - topostats version: 2.0.0
 - Python version: unknown

ns-rse, 2022-11-23:
There are a few threads in the [numpy/numpy Issues](https://github.com/numpy/numpy/issues?q=is%3Aissue+is%3Aopen+SVD+did+not+converge+in+Linear+Least+Squares) that are open.

Also a number of [StackOverflow threads that have similar problems](https://duckduckgo.com/?q=numpy+SVD+did+not+converge+in+Linear+Least+Squares).

MaxGamill-Sheffield, 2022-12-08:
I think the main problem, which is possibly more general, is that this shouldn't interrupt the processing of other files. It should have some sort of break clause and continue processing files

SylviaWhittle, 2022-12-08:
Agreed 👍

SylviaWhittle, 2023-02-16:
I should have started by looking at the image. It turns out the image is two pixels tall, which is why the polyfit fails on it.

[image]

Closing the issue and making a new one about rejecting images that are less than (10)? pixels in each direction.

closed 2023-02-16 by SylviaWhittle

MaxGamill-Sheffield, 2023-05-22:
Removed test data from drive