AFM-SPM/TopoStats #382: Statistics and image output for single molecules

Jean-Du, 2022-11-24:
**Is your feature request related to a problem? Please describe.**
At the moment, all output are either directory based or file based. An optional molecular level output is needed, especially after PR #365 is merged

**Describe the solution you'd like**
I am not sure what is the best way to approach this. With in each file-based directory in the output, we could have subdirectories named by the molecule number, eg under output/processed/examplespmfile/1, we have the cropped image, curvature plot, pixel level statistics for that one molecule.
Alternatively, we can also have directories named 'curvature' 'cropped images' etc instead of molecule number based directories.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

MaxGamill-Sheffield, 2022-11-28:
In terms of the single molecule level statistics, I think we came to the conclusion that a JSON file would be best as csv/dataframes wouldn't work well with so many statistics at different levels

MaxGamill-Sheffield, 2024-11-13:
Single object / molecule / branch etc stats either stored in own file or in the `.topostats` file

closed 2024-11-13 by MaxGamill-Sheffield