AFM-SPM/TopoStats #391: Incorrect naming of output files when input files have multiple `.`s in them.

SylviaWhittle, 2022-12-07:
**Describe the bug**

While the issue of the crashes when processing files with multiple `.`s in the file name was resolved in #271, the strings following the second `.` is lost. This second string is important for distinguishing different iterations / versions of similar scans. 

Currently, if one processes two images with the same name before the second `.`, but different after the second `.`, the second processed image will be saved with the same name as the first and so overwrite the data of the first.

Example here:

File names that differ only in post-second `.` text:
[image]

Output data:
[image]

**To Reproduce**

Run two images with the same name before the second `.`, but differing after the second `.`. Notice that the second image overwrites the output for the first image.

**Expected behavior**
The output images should retain the entire names, but have the second `.` removed. Perhaps like this:

in:
```
data.01.spm
```

out:
```
data01.png
```

**(Not required if using Docker) Your computer configuration (please complete the following information):**
 - MacOS
 - topostats version: 2.0.0
 - Python version: 3.10.

MaxGamill-Sheffield, 2022-12-08:
I remember this being an issue a while ago and I thought it was resolved. Additionally, I don't see this on the branch I'm working on (the 183 branch which hasn't touched plotting):
[image]
[image]

Are there specific circumstances where this occurs?

**Your computer configuration:**
- MacOS
-  TopoStats version: 305d481 (HEAD -> maxgamill-sheffield/183-dnatracing-refactor, origin/maxgamill-sheffield/183-dnatracing-refactor) Removed troubleshooting savetxt's
- Python: 3.10.6

SylviaWhittle, 2022-12-08:
That is very interesting. I know why it occurs on my system:

In `process_scan`, on line `170`:
```
    filename = image_path.stem
```

Now when I run topostats on the image in the screenshot, `image_path` is the file name, **without** the `.spm` extension. Now when `.stem` is called on it, it fetches the string before the `.`, but the only `.` is the one in the middle of the file name, so the name gets truncated into just the stuff before the first `.`.

This is all on the scars branch, but I wouldn't have thought that it would change between scars and main since I haven't edited anything to do with the file handling. I will check main though too now you mention it.

MaxGamill-Sheffield, 2022-12-08:
This may also be addressed in the asd PR because I add more functions to `io.py`

PR #434 (2023-01-13): Corrects image paths, lints code, adds command line options and more
Closes #362 
Closes #391 
Closes #428 
Closes #430 
Closes #433 

Pull request and code changes grew after initially making #429 (see for useful comments/feedback) but in summary...

## Double periods

`LoadScans()` class gets the filename using `Path.stem` method. However `processed_scan()` in `run_topostats.py` also took `Path.stem` so if filenames had two (or more) periods in them the filename ended up truncated and the last two sections after periods were excised (e.g. `mini.circle.spm` > `mini`). This is resolved by usin [... cut]
closed 2023-01-13 by ns-rse