AFM-SPM/TopoStats #396: Clarify install docs (paths, cwd, downloadable config file, broken link)

MaxGamill-Sheffield, 2022-12-08:
**Describe the bug**

- [ ] . - relative path clarification as some users are unsure why/what the dot is and why TS can’t find their files
- [ ] Additionally, add pwd command to show where you are in the terminal, and explain that with pip install topostats , users only need a config file and data folder
- [ ] Need to ensure that config file is downloadable or have copy-paste instructions. There is currently an issue where the config file linked in the docs and the file+validation on PyPi are different so copy pasting this gives errors
- [ ] [Readthedocs : TopoStats](https://topostats.readthedocs.io/en/dev/) (broken link) > should go to 2.0.0 docs not 1.0

**Expected behavior**
A clear and concise description of these instructions in the docs.

ns-rse, 2022-12-08:
Ah, I've already started doing most of this as part of #393.

MaxGamill-Sheffield, 2022-12-08:
@ns-rse Which have you started? I've added a few bits to on a branch but can back off?

ns-rse, 2022-12-16:
Closed by #398 

Documentation updated with...

* [x] Explanations on very basic command line usage.
* [x] Added function to generate configuration file to `run_topostats --create-config-file <filename>.yaml`
* [x] Removed broken link.

closed 2022-12-16 by ns-rse