AFM-SPM/TopoStats #401: TopoStats hangs while processing

SylviaWhittle, 2022-12-13:
**Describe the bug**
While processing the following spm file, topostats hangs at the `plotting filtering images` stage.

Spm file:
https://drive.google.com/drive/folders/1THPAuVT--iRruk2kZZWIh3ZEzkVvgyhc

**To Reproduce**

Run topostats with the following config:

```
base_dir: ./raphael/20220818/E1
output_dir: ./output/20220818/E1
warnings: ignore
cores: 4
quiet: false
file_ext: .spm
loading:
  channel: Height
filter:
  run: true # Options : true, false
  threshold_method: std_dev # Options : otsu, std_dev, absolute
  otsu_threshold_multiplier: 1.0
  threshold_std_dev:
    lower: 10.0
    upper: 1.0
  threshold_absolute:
    lower: -1.0
    upper: 1.0
  gaussian_size: 1.0121397464510862 # in px
  gaussian_mode: nearest
grains:
  run: true # Options : true, false
  # Thresholding by height
  threshold_method: std_dev # Options : std_dev, otsu, absolute
  otsu_threshold_multiplier: 1.0
  threshold_std_dev:
    lower: 10.0
    upper: 0.8
  threshold_absolute:
    lower: -1.0
    upper: 1.0
  direction: upper # Options: upper, lower, both (defines whether to look for grains above or below thresholds or both)
  # Thresholding by area
  absolute_smallest_grain_size: 100
  absolute_area_threshold:
    upper: [ 300, 100000 ] # above surface [Low, High] in nm^2 (also takes null)
    lower: [ null, null ] # below surface [Low, High] in nm^2 (also takes null)
grainstats:
  run: true # Options : true, false
  cropped_size: 40.0 # Length (in nm) of square cropped images (can take -1 for grain-sized box)
  save_cropped_grains: true # Options : true, false
dnatracing:
  run: true # Options : true, false
plotting:
  run: true # Options : true, false
  save_format: png # Options : see https://matplotlib.org/stable/api/_as_gen/matplotlib.pyplot.savefig.html
  image_set: all  # Options : all, core
  zrange: [-4, 4]  # low and high height range for core images (can take [null, null])
  colorbar: true  # Options : true, false
  axes: true # Options : true, false (due to off being a bool when parsed)
  cmap: nanoscope # Options : nanoscope, afmhot
  histogram_log_axis: false # Options : true, false
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Output**

```
❯ run_topostats --config ./my_config.yaml
[Tue, 13 Dec 2022 17:24:34] [INFO    ] [topostats] Configuration is valid.
[Tue, 13 Dec 2022 17:24:34] [INFO    ] [topostats] Plotting configuration is valid.
[Tue, 13 Dec 2022 17:24:34] [INFO    ] [topostats] Configuration file loaded from      : ./my_config.yaml
[Tue, 13 Dec 2022 17:24:34] [INFO    ] [topostats] Scanning for images in              : raphael/20220818/E1
[Tue, 13 Dec 2022 17:24:34] [INFO    ] [topostats] Output directory                    : output/20220818/E1
[Tue, 13 Dec 2022 17:24:34] [INFO    ] [topostats] Looking for images with extension   : .spm
[Tue, 13 Dec 2022 17:24:34] [INFO    ] [topostats] Images with extension .spm in raphael/20220818/E1 : 1
[Tue, 13 Dec 2022 17:24:34] [INFO    ] [topostats] Thresholding method (Filtering)     : std_dev
[Tue, 13 Dec 2022 17:24:34] [INFO    ] [topostats] Thresholding method (Grains)        : std_dev
[Tue, 13 Dec 2022 17:24:34] [INFO    ] [topostats] Extracting image from raphael/20220818/E1/E1-24h_4x_NiCl2_washes.0_00004.spm
[Tue, 13 Dec 2022 17:24:34] [INFO    ] [topostats] Loading image from : raphael/20220818/E1/E1-24h_4x_NiCl2_washes.0_00004.spm
[Tue, 13 Dec 2022 17:24:34] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes.0_00004] : Loaded image from : raphael/20220818/E1/E1-24h_4x_NiCl2_washes.0_00004.spm
[Tue, 13 Dec 2022 17:24:34] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes.0_00004] : Extracted channel Height
[Tue, 13 Dec 2022 17:24:34] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes.0_00004] : Pixel to nm scaling : 0.323564453125
Processing images from raphael/20220818/E1, results are under output/20220818/E1:   0%|               | 0/1 [00:00<?, ?it/s][Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] image_path: raphael/20220818/E1/E1-24h_4x_NiCl2_washes.0_00004
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] image_path: raphael/20220818/E1/E1-24h_4x_NiCl2_washes_0_00004
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] Processing : E1-24h_4x_NiCl2_washes_0_00004
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] NO SUFFIX : 
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] output_dir: output/20220818/E1
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] _output_dir: output/20220818/E1/Processed
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] processed path: output/20220818/E1/Processed
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] folder path: raphael/20220818/E1
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats]  filter out path: output/20220818/E1/Processed/E1-24h_4x_NiCl2_washes_0_00004/filters
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats]  grains out path: output/20220818/E1/Processed/E1-24h_4x_NiCl2_washes_0_00004/grains
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes_0_00004] : Median flattening without mask
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes_0_00004] : Plane tilt removal without mask
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] x-polyfit 1st order: [-0.09909526 27.41279646]
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] y-polyfit 1st order: [ 1.13340170e-13 -3.98712395e-11]
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] removing x plane tilt
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] removing y plane tilt
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes_0_00004] : Remove quadratic bow without mask
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] x polyfit 2nd order: [ 3.43927822e-05 -1.75747117e-02  2.89066470e+01]
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes_0_00004] : Masking (upper) Threshold: 4053817.4865054656
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes_0_00004] : Masking (lower) Threshold: -37051116.12882821
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes_0_00004] : Median flattening with mask
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes_0_00004] : Plane tilt removal with mask
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] x-polyfit 1st order: [ 6.05445577e-05 -1.45651818e-01]
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] y-polyfit 1st order: [ 5.23540105e-15 -1.79234028e-12]
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] removing x plane tilt
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] removing y plane tilt
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes_0_00004] : Remove quadratic bow with mask
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] x polyfit 2nd order: [ 2.62783013e-05 -1.34282120e-02  9.95746199e-01]
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes_0_00004] : Zero averaging background : -15644.27798071845 nm
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes_0_00004] : Applying Gaussian filter (mode : nearest; Gaussian blur (px) : 1.0121397464510862).
[Tue, 13 Dec 2022 17:24:36] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes_0_00004] : Plotting Filtering Images
```

**(Not required if using Docker) Your computer configuration (please complete the following information):**
 - OS: MacOS
 - topostats version: 2.0.0
 - Python version: 3.10

ns-rse, 2023-02-16:
This now plots with `main` (commit `0a1e028`). The resulting image is a little strange so I'm not sure if there is still a problem with processing this or it can be considered resolved as it runs to completion.

There are no grains found....

```
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes.0_00004] : Plotting Filtering Images
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [01-pixels] : Image saved to : output/processed/E1-24h_4x_NiCl2_washes.0_00004/filters/01-pixels
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [02-initial_median_flatten_unmasked] : Image saved to : output/processed/E1-24h_4x_NiCl2_washes.0_00004/filters/02-initial_median_flatten_unmasked
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [03-initial_tilt_removal_unmasked] : Image saved to : output/processed/E1-24h_4x_NiCl2_washes.0_00004/filters/03-initial_tilt_removal_unmasked
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [04-initial_quadratic_removal_unmasked] : Image saved to : output/processed/E1-24h_4x_NiCl2_washes.0_00004/filters/04-initial_quadratic_removal_unmasked
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [05-initial_scar_removal] : Image saved to : output/processed/E1-24h_4x_NiCl2_washes.0_00004/filters/05-initial_scar_removal
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [07-secondary_align_masked] : Image saved to : output/processed/E1-24h_4x_NiCl2_washes.0_00004/filters/07-secondary_align_masked
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [08-secondary_tilt_removal_masked] : Image saved to : output/processed/E1-24h_4x_NiCl2_washes.0_00004/filters/08-secondary_tilt_removal_masked
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [09-quadratic_removal_masked] : Image saved to : output/processed/E1-24h_4x_NiCl2_washes.0_00004/filters/09-quadratic_removal_masked
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [10-scar_removal] : Image saved to : output/processed/E1-24h_4x_NiCl2_washes.0_00004/filters/10-scar_removal
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [11-scar_mask] : Image saved to : output/processed/E1-24h_4x_NiCl2_washes.0_00004/filters/11-scar_mask
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [06-binary_mask] : Image saved to : output/processed/E1-24h_4x_NiCl2_washes.0_00004/filters/06-binary_mask
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [12-zero_average_background] : Image saved to : output/processed/E1-24h_4x_NiCl2_washes.0_00004/filters/12-zero_average_background
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [13-gaussian_filtered] : Image saved to : output/processed/E1-24h_4x_NiCl2_washes.0_00004/filters/13-gaussian_filtered
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes.0_00004] : Image saved to : output/processed/E1-24h_4x_NiCl2_washes.0_00004
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes.0_00004] : *** Grain Finding ***
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes.0_00004] : Thresholding method (grains) : std_dev
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes.0_00004] : Processing upper threshold (3297902.773662941)
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes.0_00004] : Masking (upper) Threshold: 3297902.773662941
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes.0_00004] : Labelling Regions
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes.0_00004] : Tidying borders
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes.0_00004] : Removing noise (upper)
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes.0_00004] : Area thresholding grains | Thresholds: L:477.58,U:{upper / self.pixel_to_nm_scaling**2:.2f} px^2, L:{lower:.2f}, U:{upper:.2f} nm^2.
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes.0_00004] : Removing small / large grains (upper)
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes.0_00004] : Area thresholding grains | Thresholds: L:2865.49,U:{upper / self.pixel_to_nm_scaling**2:.2f} px^2, L:{lower:.2f}, U:{upper:.2f} nm^2.
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes.0_00004] : Labelling Regions
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes.0_00004] : Region properties calculated (upper)
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes.0_00004] : Coloured regions
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes.0_00004] : Extracted bounding boxes (upper)
[Thu, 16 Feb 2023 11:34:13] [INFO    ] [topostats] [E1-24h_4x_NiCl2_washes.0_00004] Calculation of grainstats disabled, returning empty data frame.
Processing images from /home/neil/work/projects/topostats/tmp/401, results are under output: 100%|███████████████████████████████████████████████████| 1/1 [00:04<00:00,  4.46s/it]
[Thu, 16 Feb 2023 11:34:13] [ERROR   ] [topostats] No grains found in any images, consider adjusting your thresholds.
[Thu, 16 Feb 2023 11:34:13] [ERROR   ] [topostats] [run_topostats.py] [222] No grains found in any images, consider adjusting your thresholds.
[Thu, 16 Feb 2023 11:34:13] [ERROR   ] [topostats] All objects passed were None
[Thu, 16 Feb 2023 11:34:13] [ERROR   ] [topostats] [run_topostats.py] [223] All objects passed were None
[... cut ...]
  ^1 Successful processing of an image is detection of grains and calculation of at least
     grain statistics. If these have been disabled the percentage will be 0.

  If you encounter bugs/issues or have feature requests please report them at the above URL
  or email us.

  If you have found TopoStats useful please consider citing it. A Citation File Format is
  linked above and available from the Source Code page.
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

...but the Filtering images are plotted.

[image]

SylviaWhittle, 2023-02-16:
Ah. I think this data is a dud. The dataset that it was part of had not been checked for bad data.

ns-rse, 2023-02-16:
Maybe it would be worth thinking about how we check for bad data and whether it might become part of the TopoStats workflow.  No idea how this is currently done though.

closed 2023-02-16 by ns-rse

MaxGamill-Sheffield, 2023-05-22:
Removed drive test data