AFM-SPM/TopoStats #402: Dnatracing division by zero

SylviaWhittle, 2022-12-13:
**Describe the bug**
When processing the following file, TopoStats throws a division by zero error during `dnatracing`. 

**To Reproduce**

Run topostats on the following file and with the following config:

File: https://drive.google.com/drive/folders/1mFtdYHl6g0kxxBpJcpY9a65t1TP-kqyU

Config:

```
base_dir: ./raphael/20220819/C3
output_dir: ./output/20220819/C3
warnings: ignore
cores: 4
quiet: false
file_ext: .spm
loading:
  channel: Height
filter:
  run: true # Options : true, false
  threshold_method: std_dev # Options : otsu, std_dev, absolute
  otsu_threshold_multiplier: 1.0
  threshold_std_dev:
    lower: 10.0
    upper: 1.0
  threshold_absolute:
    lower: -1.0
    upper: 1.0
  gaussian_size: 1.0121397464510862 # in px
  gaussian_mode: nearest
grains:
  run: true # Options : true, false
  # Thresholding by height
  threshold_method: std_dev # Options : std_dev, otsu, absolute
  otsu_threshold_multiplier: 1.0
  threshold_std_dev:
    lower: 10.0
    upper: 1.0
  threshold_absolute:
    lower: -1.0
    upper: 1.0
  direction: upper # Options: upper, lower, both (defines whether to look for grains above or below thresholds or both)
  # Thresholding by area
  absolute_smallest_grain_size: 100
  absolute_area_threshold:
    upper: [ 300, 100000 ] # above surface [Low, High] in nm^2 (also takes null)
    lower: [ null, null ] # below surface [Low, High] in nm^2 (also takes null)
grainstats:
  run: true # Options : true, false
  cropped_size: 40.0 # Length (in nm) of square cropped images (can take -1 for grain-sized box)
  save_cropped_grains: true # Options : true, false
dnatracing:
  run: true # Options : true, false
plotting:
  run: true # Options : true, false
  save_format: png # Options : see https://matplotlib.org/stable/api/_as_gen/matplotlib.pyplot.savefig.html
  image_set: all  # Options : all, core
  zrange: [-4, 4]  # low and high height range for core images (can take [null, null])
  colorbar: true  # Options : true, false
  axes: true # Options : true, false (due to off being a bool when parsed)
  cmap: nanoscope # Options : nanoscope, afmhot
  histogram_log_axis: false # Options : true, false
```

**Output**

```
❯ run_topostats --config ./my_config.yaml
[... 72 lines cut ...]
  File "/Users/sylvi/Documents/GitKraken/TopoStats/topostats/run_topostats.py", line 352, in process_scan
    dna_traces[direction].trace_dna()
  File "/Users/sylvi/Documents/GitKraken/TopoStats/topostats/tracing/dnatracing.py", line 91, in trace_dna
    self.get_splined_traces()
  File "/Users/sylvi/Documents/GitKraken/TopoStats/topostats/tracing/dnatracing.py", line 426, in get_splined_traces
    while nbr / step_size < 4:
ZeroDivisionError: division by zero
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/Users/sylvi/opt/anaconda3/envs/scars/bin/run_topostats", line 8, in <module>
    sys.exit(main())
  File "/Users/sylvi/Documents/GitKraken/TopoStats/topostats/run_topostats.py", line 459, in main
    for img, result in pool.imap_unordered(
  File "/Users/sylvi/opt/anaconda3/envs/scars/lib/python3.10/multiprocessing/pool.py", line 873, in next
    raise value
ZeroDivisionError: division by zero
```

**(Not required if using Docker) Your computer configuration (please complete the following information):**
 - OS: MacOS
 - topostats version: 2.0.0
 - Python version: 3.10

ns-rse, 2023-02-13:
There were two files in the directory...

```
(topostats) ❱ l data 
drwxr-xr-x neil neil 4.0 KB Mon Feb 13 16:53:21 2023  .
drwxr-xr-x neil neil 4.0 KB Mon Feb 13 16:51:42 2023  ..
.rw-r--r-- neil neil 591 KB Mon Feb 13 16:48:58 2023  C3_1ng_Imob_in_MgCl2_3x_NiCl2_washes.0_00013.spm
.rw-r--r-- neil neil 2.1 MB Mon Feb 13 16:49:02 2023  E4-24h-nm.0_00006.spm
```

Neither appear to now cause the `ZeroDivisionError` noted above.

Instead we now have...

## E4-24h-nm.0_00006

29 grains detected which are all too small for DNA tracing...

```
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : *** Grain Statistics ***
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 0
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Skipping grain due to being too small (size: (2, 3)) to calculate stats for.
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 1
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Skipping grain due to being too small (size: (3, 4)) to calculate stats for.
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 2
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Skipping grain due to being too small (size: (4, 3)) to calculate stats for.
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 3
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Skipping grain due to being too small (size: (3, 2)) to calculate stats for.
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 4
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 5
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Skipping grain due to being too small (size: (3, 4)) to calculate stats for.
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 6
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 7
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Skipping grain due to being too small (size: (3, 3)) to calculate stats for.
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 8
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Skipping grain due to being too small (size: (4, 2)) to calculate stats for.
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 9
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 10
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Skipping grain due to being too small (size: (3, 3)) to calculate stats for.
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 11
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Skipping grain due to being too small (size: (4, 6)) to calculate stats for.
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 12
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Skipping grain due to being too small (size: (4, 7)) to calculate stats for.
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 13
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Skipping grain due to being too small (size: (4, 7)) to calculate stats for.
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 14
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Skipping grain due to being too small (size: (3, 5)) to calculate stats for.
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 15
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 16
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Skipping grain due to being too small (size: (3, 3)) to calculate stats for.
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 17
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Skipping grain due to being too small (size: (3, 2)) to calculate stats for.
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 18
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Skipping grain due to being too small (size: (2, 3)) to calculate stats for.
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 19
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Skipping grain due to being too small (size: (3, 4)) to calculate stats for.
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 20
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Skipping grain due to being too small (size: (4, 2)) to calculate stats for.
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 21
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Skipping grain due to being too small (size: (2, 5)) to calculate stats for.
[Mon, 13 Feb 2023 16:51:45] [INFO    ] [topostats] [E4-24h-nm.0_00006] : Processing grain: 22
[... cut ...]
  File "/home/neil/.virtualenvs/topostats/lib/python3.10/site-packages/matplotlib/axis.py", line 1112, in _set_lim
    v0 = self.axes._validate_converted_limits(v0, self.convert_units)
  File "/home/neil/.virtualenvs/topostats/lib/python3.10/site-packages/matplotlib/axes/_base.py", line 3606, in _validate_converted_limits
    raise ValueError("Axis limits cannot be NaN or Inf")
ValueError: Axis limits cannot be NaN or Inf
```

Whether this is accurate I'm unsure, the image is...

[image]

I can fix this error too so that images with zero grains detected do not attempt to plot distributions.

closed 2023-02-16 by ns-rse

MaxGamill-Sheffield, 2023-05-22:
Removed gdrive test data

commit 613fc93e (2023-02-16): Resolving errors with empty data frames

Closes #402

Still get errors with DNA tracing of this image but it runs to completion.

Changes include...

* Changes to where indexes are set and ensuring `image` is part of pd.DataFrame index.
* Return grainstats if tracingdna fails or `dnatracing_config["run"] == False`.
* Improves flexibility of `create_empty_dataframe()` allowing index to be set explicitly.
* Only attempt to plot summary distribution if statistic have been calculated (required because configs may request
  plotting of statistics from DNA tracing which is more likely to fail, yet w [... cut]
PR #469 (2023-02-16): Resolving errors with empty data frames
Closes #402

Still get errors with DNA tracing of this image but it runs to completion.

Changes include...

* Changes to where indexes are set and ensuring `image` is part of pd.DataFrame index.
* Return `grainstats_df` if `tracingdna` fails or `dnatracing_config["run"] == False`.
* Improves flexibility of `create_empty_dataframe()` allowing index to be set explicitly.
* Only attempt to plot summary distribution if statistic has been calculated (required because configs may request plotting of statistics from DNA tracing which is more likely to fail, ye [... cut]