AFM-SPM/TopoStats #403: Add molecule number in the all_statistics.csv spreadsheet

Jean-Du, 2022-12-14:
**Is your feature request related to a problem? Please describe.**
Introduce a column of molecule number in the all_statistics.csv spreadsheet and make sure it is consistent with the molecule number in the cropped out images of all the grains, so that the user can look up the statistics for each individual grain.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

SylviaWhittle, 2022-12-14:
I think this is a reasonable and sensible feature to add. Can't see any issues with it, will help with debugging too.

ns-rse, 2022-12-16:
I thought I had ensured these were included as I'm sure I've added `.to_csv(<filename>, index=True)` option in the past.

Jean-Du, 2023-05-11:
It seems that there is still a mismatch between the molecule number in `dnatracing.py` and the molecule number in the statistics output. @MaxGamill-Sheffield has tested this with `minicircles.spm` by increasing the `purge_obvious_crap` threshold to 60 and has prodcued the following files:
[no_lils.csv](https://github.com/AFM-SPM/TopoStats/files/11454511/no_lils.csv)
[lils.csv](https://github.com/AFM-SPM/TopoStats/files/11454512/lils.csv)
[image]
[image]

ns-rse, 2023-05-12:
I think I've sussed this out but I've a question...

If a grain has statistics but is removed because of size during `dnaTrace` processing (aka `purge_obvious_crap`) is it desirable to have the grain statistics reported but no tracing statistics?

Jean-Du, 2023-05-12:
Hi @ns-rse , yes, I believe that would be desirable

commit 38871c54 (2023-02-02): Fixes #447 extra column in CSV

Closes #447

* Fixes additional column in CSV output when images have no grains detected. This was caused by the empty dummy data
  frame not having the index set to `molecule_number` (nee `Molecule Number`) and so when concatenated via `pd.concat()`
  an additional column was introduced.
* Moves `utils.folder_grainstats()` to `io.folder_grainstats()`.
* Consistent case/format for variable names in CSV (all lower snake-case).
* CSVs now include `molecule_number` Closes #403

This will fail tests because it impacts regression tests. I've recent switched to using  [... cut]
PR #562 (2023-05-16): Corrects alignment/merging of grainstats and tracing_stats dataframes
Closes #403

If DNA tracing removed grains because the skeleton was less than a hard coded threshold (10) in size then no tracing statistics were calculated for that grain (as it was dropped).

This threshold is now configurable (via `crap_threshold` since its the `purge_obvious_crap()` method that performs this).

The way molecules are currently numbered always starts at `1` because the _whole_ image with grains represented as masks is taken as input. `0` is the background and therefore removed from the molecule numbers whi [... cut]
commit 4131b4a4 (2023-05-16): Corrects alignment/merging of grainstats and tracing_stats dataframes

Closes #403

If DNA tracing removed grains because the skeleton was less than a hard coded threshold (10) in size then no tracing
statistics were calculated for that grain (as it was dropped).

This threshold is now configurable (via `crap_threshold` since its the `purge_obvious_crap()` method that performs
this).

The way molecules are currently numbered always starts at `1` because the _whole_ image with grains represented as masks
is taken as input. `0` is the background and therefore removed from the molecule numbers wh [... cut]
closed 2023-05-16 by ns-rse