AFM-SPM/TopoStats #404: Grainstats crash

SylviaWhittle, 2022-12-14:
**Describe the bug**
A crash occurs while processing an AFM image which appears to originate from `grainstats.py`. The error is `ValueError: operands could not be broadcast together with shapes (0,) (2,) ` which is an issue with handling arrays. Check grainstats code to ensure that there are measures to check for bad array size combinations and also diagnose how they can arise. 

**To Reproduce**

Run TopoStats on the following file: https://drive.google.com/drive/folders/1_EUP6WyfsU5uSTO0eUXh1JUkxdsXDEab

Using the following config:

```
base_dir: ./raphael/20220819/E4
output_dir: ./output/20220819/E4
warnings: ignore
cores: 4
quiet: false
file_ext: .spm
loading:
  channel: Height
filter:
  run: true # Options : true, false
  threshold_method: std_dev # Options : otsu, std_dev, absolute
  otsu_threshold_multiplier: 1.0
  threshold_std_dev:
    lower: 10.0
    upper: 1.0
  threshold_absolute:
    lower: -1.0
    upper: 1.0
  gaussian_size: 1.0121397464510862 # in px
  gaussian_mode: nearest
grains:
  run: true # Options : true, false
  # Thresholding by height
  threshold_method: std_dev # Options : std_dev, otsu, absolute
  otsu_threshold_multiplier: 1.0
  threshold_std_dev:
    lower: 10.0
    upper: 1.0
  threshold_absolute:
    lower: -1.0
    upper: 1.0
  direction: upper # Options: upper, lower, both (defines whether to look for grains above or below thresholds or both)
  # Thresholding by area
  absolute_smallest_grain_size: 100
  absolute_area_threshold:
    upper: [ 300, 100000 ] # above surface [Low, High] in nm^2 (also takes null)
    lower: [ null, null ] # below surface [Low, High] in nm^2 (also takes null)
grainstats:
  run: true # Options : true, false
  cropped_size: 40.0 # Length (in nm) of square cropped images (can take -1 for grain-sized box)
  save_cropped_grains: true # Options : true, false
dnatracing:
  run: true # Options : true, false
plotting:
  run: true # Options : true, false
  save_format: png # Options : see https://matplotlib.org/stable/api/_as_gen/matplotlib.pyplot.savefig.html
  image_set: all  # Options : all, core
  zrange: [-4, 4]  # low and high height range for core images (can take [null, null])
  colorbar: true  # Options : true, false
  axes: true # Options : true, false (due to off being a bool when parsed)
  cmap: nanoscope # Options : nanoscope, afmhot
  histogram_log_axis: false # Options : true, false

```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Output**

[... 85 lines cut ...]
  File "/Users/sylvi/Documents/GitKraken/TopoStats/topostats/grainstats.py", line 218, in calculate_stats
    radius_stats = self.calculate_radius_stats(edges, points)
  File "/Users/sylvi/Documents/GitKraken/TopoStats/topostats/grainstats.py", line 368, in calculate_radius_stats
    displacements = self._calculate_displacement(edges, centroid)
  File "/Users/sylvi/Documents/GitKraken/TopoStats/topostats/grainstats.py", line 400, in _calculate_displacement
    return np.array(edges) - centroid
ValueError: operands could not be broadcast together with shapes (0,) (2,) 
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/Users/sylvi/opt/anaconda3/envs/scars/bin/run_topostats", line 8, in <module>
    sys.exit(main())
  File "/Users/sylvi/Documents/GitKraken/TopoStats/topostats/run_topostats.py", line 459, in main
    for img, result in pool.imap_unordered(
  File "/Users/sylvi/opt/anaconda3/envs/scars/lib/python3.10/multiprocessing/pool.py", line 873, in next
    raise value
ValueError: operands could not be broadcast together with shapes (0,) (2,) 
```

**(Not required if using Docker) Your computer configuration (please complete the following information):**
 - OS: MacOS
 - topostats version: 2.0.0
 - Python version: 3.10

SylviaWhittle, 2023-01-09:
Cannot seem to reproduce despite using the correct config and image.

MaxGamill-Sheffield, 2023-02-06:
For me, this occurred as a user error... I had been messing around applying a gaussian filter in the grains stage on the mask. This ended up causing some grains to recced and become tiny after the area filtering stage, allowing some to seep through. It seemed as though these tiny grains (of only a few pixels) produced an `edges` array (in `calulate_edges()`) which was empty.

Minimal reproducible example:
```
from skimage.feature import canny
import numpy as np

a = np.asarray([[0,0,0,0,0],
                [0,1,1,1,0],
                [0,0,1,1,0],
                [0,0,0,1,0],
                [0,0,0,0,0]])
b = canny(a, sigma=1)
print(b)
```
See that b is all false showing there are no edges found by the canny function.

I believe this to be due to the gaussian applied to these small grains which completely removes them before applying the sobel filters as part of the canny edge detection algorithm. Perhaps we need to recheck the sizes / edges exist in the grain and discard before moving on?

SylviaWhittle, 2023-02-07:
Can coroborate that during `calculate_edges()`, the canny edge detection fails for small grains:

[image]

Making this grain a little bigger, fixes it:

[image]

SylviaWhittle, 2023-02-07:
Proposal 1:
To ensure that the size of the grain is large enough such that issues of detecting edges do not arise

This might be done by adding the following code to `grainstats.py`:
```
            # Skip grain if too small to calculate stats for
            LOGGER.info(f"[{self.image_name}] : Grain size: {region.image.size}")
            print(region.image.astype(int))
            if region.image.shape[0] > 5 and region.image.shape[1] > 5:
                LOGGER.info(f"[{self.image_name}] : Skipping grain due to being too small to calculate stats for.")
                continue
```

Which would skip grains that are not at least 5 x 5.

SylviaWhittle, 2023-02-07:
Proposal 2:
Update the edge finding method to be a little more robust.

I have written the following code that I think might be better for calculating the edges of a binary grain image.

```
# Add padding (needed for erosion)
padded = np.pad(image, 1)
print(f"padded:\n{padded}")

# Erode
eroded = skimage_morphology.erosion(padded)
print(f"eroded:\n{eroded.astype(int)}")
# Remove padding
eroded = eroded[1:-1, 1:-1]
print(f"eroded (without padding):\n{eroded.astype(int)}")

# Edges is equal to the difference between the
# original image and the eroded image.
edges = image - eroded
print(f"edges:\n{edges}")
```

This seems to work better for grains even when really small:
[image]

[image]

[image]

What do people think?

ns-rse, 2023-02-07:
I like the enhancement to edge detection, its vectorised and elegant.

But I'd perhaps suggest going with both since removing small "things" is something that is being done at various stages during grain detection and later during DNA Tracing. I like cleaning things up as early in a workflow as possible as it means subsequent steps don't need to have so many clauses (e.g. the current `purge_obvious_crap()` method in DNA tracing).

If we can remove "noise" as early as possible that is a good thing.

Another way of doing the test in proposal 1 that is fractionally more succinct might be

```python
if max(region.image.shape) < 5:
                LOGGER.info(f"[{self.image_name}] : Skipping grain due to being too small to calculate stats for.")
                continue
```

I've also not put it into context but presumably what follows are steps for edge detection and the building of the grains objects which are then used. Just need to make sure that `region.image.shape` doesn't get included.

closed 2023-02-13 by SylviaWhittle

MaxGamill-Sheffield, 2023-05-22:
Removed gdrive test data

PR #454 (2023-02-13): Sylvia whittle/404 grainstats crash
Tiny PR! 

Closes #404 

This PR aims to fix bug #404, by both preventing tiny grains (less than 5x5) being processed in grainstats, and also to improve the edge finding algorithm so that if in future we want to handle grains less than 5x5, `calculate_edges` will be able to handle this. (Note that other functions may still have issue with this). So in effect, this *should* be both prevention and cure for the bug. 

I have verified that this works locally on my machine, however for you to verify that it fixes the bug, run latest `main` TopoStats on this file: [... cut]