AFM-SPM/TopoStats #406: Skeletons appear broken when plotting

MaxGamill-Sheffield, 2022-12-16:
**Describe the bug**
Single pixel thick skeletons appear broken on the produced plots, this seems to be due to the reduced dpi.

**To Reproduce**
This has been made to break-up the knots and cats branch. I have obtained skeletons from dnatracing, and used the skeleton as a mask for the `data2` input to `plottingfuncs.Images`

**Expected behavior**
Skeletons should be visible:
[image]

**Output**
Skeletons seem broken:
[image]

ns-rse, 2022-12-16:
What version/branch/commit are you using? Is this TopoStats 2.0.0 or another branch?

If its that with refactoring of DNATracing that you've been working on which method of skeletonisation are you using? Is it "Joe's" with the pruning and could that be a factor? Is the expected image above a single-pixel plot and if so how was it skeletonised?

MaxGamill-Sheffield, 2022-12-19:
This is a dpi error and fixed with #409 (pending weird test error resolution)

PR #409 (2023-01-17): Maxgamill sheffield/406 plot improvements
Fixes #406
This PR adds:
- dpi and mask_cmap into plottingfuncs (the dpi setting will be applied on the plottingdict level and not be an adjustable parameter in the config - mainly because this is useful for skeletons and not much else)
- gets the cmaps in `init` instead of in the code for readability
- adds `gwyddion` and `blu` colourmaps
- adds mask_cmap to default config
- Adds these to validation
- Adds single test using small array and mask for the 'blu' colourmap

Example with `blu` "mask_cmap" selected:
[image]
closed 2023-01-17 by MaxGamill-Sheffield