AFM-SPM/TopoStats #412: dnatracing refactor - skeletonisation tests required

MaxGamill-Sheffield, 2022-12-20:
(branch: maxgamill-sheffield/183-dnatracing-refactor)
This branch centres around refactoring dnatracing.py upto and including the disrodered_traces function. This means this PR centers around organising the skeletonisation code from tracingfuncs, removing the unused code, adding docstrings and putting it into skeletonize.py.

The regtests remain unaltered and pass so things are working as expected, some dnatracing tests have been made, but no additional tests have been made for Joe's skeletonisation code as I'm unsure what some of them do, there is a lot, and the class is meant to act like Zhangs method (but produces different results). - If anyone can offer assistance here, that would be very helpful!

I also altered some of the utility functions in dnatracing.py to be more understandable. However, I do have an issue with a bit of the code - dilation requires each grain to have their own binary image (to prevent close grains touching), but this also dramatically slows processing times and workflow as the whole image can't be skeletonised/pruned at once (which all the skeletonisation methods can do).

ns-rse, 2023-01-10:
I've undertaken refactoring `dnatracing.py` on `ns-rse/183a-dnatracing-refactor` and am abstracting out the processing of the whole image so that single grains are the unit of processing for DNA tracing.

Tests are being written and the Skeletonisation (and in due course ridge detection) methods from Scikit Image are being incorporated.

I'll look at the dilation issue as I've used the `scipy.ndimage.binary_dilation()` function which may already be faster (not had time/inclination to check but am vectorising the linear/circular detection rather than current conversion to lists and looping and will be timing that).

MaxGamill-Sheffield, 2023-01-10:
I think @SylviaWhittle has actually written tests for all of joes functions on a branch. Sylvia, was this on my branch or another?

SylviaWhittle, 2023-01-10:
@ns-rse @MaxGamill-Sheffield I wrote tests for skeletonization in `maxgamill-sheffield/183-dnatracing-refactor`, though IIRC there was one function where I needed more information on intended behaviour before writing a test. Also I don't want to claim that these tests are in a publish-ready form.

SylviaWhittle, 2023-01-10:
(Pushed just now - realised I hadn't pushed 😨 )

ns-rse, 2024-11-26:
There are some unit tests in place but most of the classes methods are awaiting being written.