AFM-SPM/TopoStats #423: Absolute values : validation failing

SylviaWhittle, 2023-01-09:
**Describe the bug**
As reported by @MaxGamill-Sheffield, the validation is failing for valid values of absolute threshold values. 

**To Reproduce**

Run topostats with a variety of absolute thresholds, there will be some / all that are invalid incorrectly. 

**Suggestions** 

@MaxGamill-Sheffield suggests that once this has been amended, we could also do zeroing steps so that the absolute thresholds are not just relative to the mean pixel value for the image.

ns-rse, 2023-01-09:
This ties in with #302 and the need to discuss/know what sensible values are for _all_ fields in configuration files so that the validation fields and ranges against which they are validated can be set sensibly.

ns-rse, 2023-01-09:
Also which version of TopoStats was this found to fail with? Is it `v2.0.0` that users have installed from PyPI or were they installing from `main` branch on GitHub?

SylviaWhittle, 2023-01-09:
@MaxGamill-Sheffield any further info like @ns-rse is suggesting above would be great if you can recall it

ns-rse, 2023-01-09:
Another question...there are two sets of `threshold_absolute_[lower|upper]`, one under `filter:` and one under `grains:` do we know which was causing errors?

Current validation fields for `v2.0.0` are (from `topostats/validation.py`)...

`filter.threshold_absolute.lower < 0`
`filter.threshold_absolute.upper > 0`
`grains.threshold_absolute.lower < 0`
`grains.threshold_absolute.upper > 0`

So there is a fairly clear case of what is going to cause failures. The most likely I suspect is going to be a configuration option where absolute_lower that is >= 0.

MaxGamill-Sheffield, 2023-01-09:
So I believe this was on the PyPI release but when checking my own validation it was present there too, and here we are interested in the Filters class but it may also be applicable to the Grains class.

Essentially, the results that come out before obtaining a mask can differ greatly. The histogram below shows a small range compared to some data I saw today which had this peak at -10nm and a "second bump" we were interested in at ~ -4nm. However, the requirement for the upper to be > 0 stopped this.
[image]

As Sylvia mentioned before, this depends on the data being processed and therefore none of the values will match between images as the means may be moved about due to impurities / higher molecule concentrations etc, making it hard to find our "surface" every time. However, if (when absolute was chosen) we could shift the data so its min values were around 0, these numbers would make more sense, and be more generalisable.

PR #426 (2023-01-10): Relaxes validation of absolute_threshold
Closes #423

I understand the problem but do not feel this is an optimal solution given the situation that these absolute thresholds are image specific and the aim to use TopoStats to batch process images.

There are two threads here...

1) The need to have sensible values against which validation is performed as noted for discussion in #302.

2) The challenge of trying to write software and have defaults that cover all scenarios (or say ~80%).

Given the threshold can change on a per-image basis I do not think having user specified `absolute_threshold` [... cut]
commit 9d625130 (2023-01-10): Relaxes validation of absolute_threshold

Closes #423

I understand the problem but do not feel this is an optimal solution given the situation that these absolute thresholds
are image specific and the aim to use TopoStats to batch process images.

There are two threads here...

1) The need to have sensible values against which validation is performed as noted for discussion in #302.

2) The challenge of trying to write software and have defaults that cover all scenarios (or say ~80%).

Given the threshold can change on a per-image basis I do not think having user specified `absolute_threshold [... cut]
closed 2023-01-10 by ns-rse