AFM-SPM/TopoStats #427: Logging throwing OSError

ns-rse, 2023-01-10:
**Describe the bug**

Logging is creating errors in some random instances when batch processing images.

In an attempt to debug the number of `cores: 1` has been attempted and the error did not occur.

The following configuration file was used...

```
base_dir: ./
output_dir: ./output_ba
warnings: ignore
cores: 4
quiet: false
file_ext: .spm
loading:
  channel: Height
filter:
  run: true # Options : true, false
  threshold_method: std_dev # Options : std_dev (default), otsu, absolute
  otsu_threshold_multiplier: 1.0
  threshold_std_dev:
    lower: 10.0
    upper: 1.0
  threshold_absolute:
    lower: -1.0
    upper: 1.0
  gaussian_size: 1.0121397464510862 # in px
  gaussian_mode: nearest
grains:
  run: true # Options : true, false
  # Thresholding by height
  threshold_method: std_dev # Options : std_dev, otsu, absolute
  otsu_threshold_multiplier: 1.0
  threshold_std_dev:
    lower: 1.0
    upper: 1.0
  threshold_absolute:
    lower: -1.0
    upper: 1.0
  direction: upper # Options: upper, lower, both (defines whether to look for grains above or below thresholds or both)
  # Thresholding by area
  absolute_smallest_grain_size: 100
  absolute_area_threshold:
    upper: [ 300,800 ] # above surface [Low, High] in nm^2 (also takes null)
    lower: [ null, null ] # below surface [Low, High] in nm^2 (also takes null)
grainstats:
  run: true # Options : true, false
  cropped_size: 60.0 # Length (in nm) of square cropped images (can take -1 for grain-sized box)
  save_cropped_grains: true # Options : true, false
dnatracing:
  run: true # Options : true, false
plotting:
  run: true # Options : true, false
  save_format: png # Options : see https://matplotlib.org/stable/api/_as_gen/matplotlib.pyplot.savefig.html
  image_set: all  # Options : all, core
  zrange: [0, 4]  # low and high height range for core images (can take [null, null])
  colorbar: true  # Options : true, false
  axes: true # Options : true, false (due to off being a bool when parsed)
  cmap: nanoscope # Options : nanoscope, afmhot
  histogram_log_axis: false # Options : true, false
```

**To Reproduce**

Sample files causing this error will be linked in comments.

**Expected behavior**

Images should process when run in parallel.
[... 666 lines cut ...]
  File "C:\Users\Mingxue\anaconda3\envs\TopoStats\lib\multiprocessing\spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "C:\Users\Mingxue\anaconda3\envs\TopoStats\lib\multiprocessing\spawn.py", line 129, in _main
    return self._bootstrap(parent_sentinel)
  File "C:\Users\Mingxue\anaconda3\envs\TopoStats\lib\multiprocessing\process.py", line 315, in _bootstrap
    self.run()
  File "C:\Users\Mingxue\anaconda3\envs\TopoStats\lib\multiprocessing\process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\Mingxue\anaconda3\envs\TopoStats\lib\multiprocessing\pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "C:\Users\Mingxue\PycharmProjects\TopoStats\topostats\run_topostats.py", line 197, in process_scan
    filtered_image.filter_image()
  File "C:\Users\Mingxue\PycharmProjects\TopoStats\topostats\filters.py", line 340, in filter_image
    self.images["masked_quadratic_removal"] = self.remove_quadratic(
  File "C:\Users\Mingxue\PycharmProjects\TopoStats\topostats\filters.py", line 211, in remove_quadratic
    LOGGER.info(f"[{img_name}] : Remove quadratic bow with mask")
Message: '[20221028_HU_1.ng_DDNA_7.5ng_PLL.0_00006] : Remove quadratic bow with mask'
Arguments: ()
[Mon, 09 Jan 2023 18:39:37] [INFO    ] [topostats] x polyfit 2nd order: [ 6.04224491e-07 -6.18121654e-04  1.25660438e-01]
```

**(Not required if using Docker) Your computer configuration (please complete the following information):**
 - OS: Windows
 - topostats version: Branch `JeanDu/BendingAngle`; commit `38b5f4b`
 - Python version: 3.10

Jean-Du, 2023-01-10:
Example files that led to this error: https://drive.google.com/drive/folders/1svlyXdjjMzrMjiOBLIZLPcQesUReRfsh
When each file was processed on its own, such error did not occur, but it occured when two or more files are processed together

MaxGamill-Sheffield, 2023-01-10:
Tried on the main branch and get a "Processed" directory not there error (and a "Processed" directory is produced but not in "output", but once resolved I don't see this error on the main branch.

Jean-Du, 2023-01-10:
Update: further testing shows that the error doesn't seem to be associated with any specific file. I tried replacing the files with `minicircle.spm` and[ `339bp.spm`](https://drive.google.com/file/d/1w9nUbDIHBaMAdMmNRbkP1xJxWbYcf9d1/view?usp=share_link), and still got the same error

Jean-Du, 2023-01-10:
I forgot to mention that lines 339-351 in `run_topostats.py` need to be commented off from the branch, as they were only used for testing one file.

Jean-Du, 2023-01-10:
It seems that this error is related to the loggers I added in `measure_bending_angle` in `dnatracing.py`, and commenting out those loggers resolves the error.