AFM-SPM/TopoStats #428: Images saved to wrong place & Processed dir error

MaxGamill-Sheffield, 2023-01-10:
**Describe the bug**
No matter where the output_dir is set, all images go into a `./Processed` folder that is made instead of `./<output_dir>/Processed`. This is an issue because the "`older_statistics.csv` file does want to go into the latter, throwing an error when this dir doesn't exist.

**To Reproduce**
Run topostats (main branch)
See "Processed" directory in `.`
Error thrown due to missing 

**Expected behavior**
Processed folder should be in "./<output_dir>"

**Output**
Something along the lines of :
`OSError: "Processed"  directory not found`

**Additional context**
FIX:
Remove the `.parent` on line 173 -> ` _output_dir = get_out_path(image_path, base_dir, output_dir) / "Processed"`

slackline, 2023-01-10:
Strange, that `.parent` call hasn't changed in five months since you implemented it, can't think why it would suddenly break. :thinking:

MaxGamill-Sheffield, 2023-01-10:
Same, and I'm guessing it's the same for the function beneath it too

ns-rse, 2023-01-10:
Apologies commented from the wrong browser where I am logged into my personal GitHub account.

One thought is perhaps related to OS, is this happening on Windows or across all tested platforms?

Directory creation is currently a bit messy I feel. I'd started with a user specified `output_dir` and then a sub-directory for each image. The introduction of images and making the distinction of `processed` introduced another layer as did splitting images into `filter` and `grains` and then a further layer with `upper` and `lower`.

Perhaps taking a step back and reviewing what is essential and the simplest structure we can deduce might be appropriate rather than putting a plaster on this.

Jean-Du, 2023-01-10:
I believe this is caused by periods in the file name. 
In `get_out_path` in `utils.py`, lines 100-111 treat image paths with suffixes differently from those without suffixes, and the parent directory is used for the former case. For images with periods in their names, the part after the period is for some reason wrongly treated as a suffix, causing the creating of `processed` in the wrong place.
There are two ways to prove such is the case:
1. Remove all periods from the file name, and the `processed` folder should be in the right place
2. For a file that does contain periods, remove the `parent` part from lines 104 and/or 109 in `utils.py`, and the `processed` folder should be in the right place

SylviaWhittle, 2023-01-10:
I tried this PR out on my system with a variety of different folder setups, and it seems to do as @ns-rse says. It does not fix the double periods issue (though this PR isn't intending to AFAIK - I checked just in case and to clarify), so that still needs a PR after this one.

Here is an example of the file structure, could someone verify if this is intended? Looks good to me.

[image]

PR #434 (2023-01-13): Corrects image paths, lints code, adds command line options and more
Closes #362 
Closes #391 
Closes #428 
Closes #430 
Closes #433 

Pull request and code changes grew after initially making #429 (see for useful comments/feedback) but in summary...

## Double periods

`LoadScans()` class gets the filename using `Path.stem` method. However `processed_scan()` in `run_topostats.py` also took `Path.stem` so if filenames had two (or more) periods in them the filename ended up truncated and the last two sections after periods were excised (e.g. `mini.circle.spm` > `mini`). This is resolved by usin [... cut]
commit 18ef6909 (2023-01-13): Corrects path output construction in get_out_path

Closes #428

Some minor linting errors are also corrected (but not all resolved as one has been disabled).
closed 2023-01-13 by ns-rse