AFM-SPM/TopoStats #431: Remove grains based on height

Jean-Du, 2023-01-12:
**Is your feature request related to a problem? Please describe.**
It would be useful to be able to remove grains for being too high. This could prevent certain impurities having the same area as molecules of interest from being selected.

**Describe the solution you'd like**
Introduce another user configurable threshold based on the mean height of the grain, so that for example, grains whose mean height deviates from the image mean by over 5 sigma can be excluded. (Suggested by @SylviaWhittle ) 
Example images will be supplied as google drive links in the comments.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Jean-Du, 2023-01-12:
As an example, this [spm file](https://drive.google.com/file/d/1qWwXFDSUpAy_slZ_xKA7BS6KTeXQkya6/view?usp=share_link) contains grains that are too high and aren't of interest (see [processed](https://drive.google.com/file/d/1qYxncHLmXA0GOuoOVdeeGcfifM-7MDOs/view?usp=share_link) and [masked](https://drive.google.com/file/d/1H2XZDdR7xf-tDUfjnZQjigO7E7MYyBv0/view?usp=share_link) images).
If this feature request is implemented, grains circled in blue in [this image](https://drive.google.com/file/d/1qXc-YbZg6-bQVpzn4CE8_YeDWyjMx_zO/view?usp=share_link) can be excluded for being too high.

ns-rse, 2023-01-16:
AS #390 closed #154 I think this can be closed too as its a duplicate.

closed 2023-01-16 by ns-rse

Jean-Du, 2023-01-16:
@ns-rse Hi, I think this feature request is slight different from #154  as it's about user configurable grain removal in the grains section, whereas the scar removal is in the filters section and doesn't remove globular features, such as those shown in the example images above. @SylviaWhittle should be able to confirm this.

ns-rse, 2023-04-21:
If the overall aim of developing TopoStats is to provide a "hands-free" experience (as much as that it is possible) where output is as good as it can be I feel it is appropriate to  take time to consider and implement features and base it on empirical data to justify the design decisions rather than just adding options just because we can.

This would require systematic assessment of a wide range of scans from different sample types (not just `minicircles.spm`)  to provide the data on which to base such decisions.

I'd also question what to do when such regions are removed, as it would result in "blank" regions as its not possible to interpolate what is behind the high regions. If there are other grains of interest going into such a region you can't really extrapolte/interpolate what would be behind so you're left with a "hole" in the image. Is that any better than the existing high region? To my crude understanding they look like impurities in the sample/buffer mixture e.g. dirt. Sometimes the answer to such issues isn't algorithmic, but rather better data.

Jean-Du, 2023-04-21:
Hi @ns-rse , thanks for your thoughts! I would like to clarify that removing those grains would not lead to blank regions, as this step would happen in 'grains' and not 'filter': the idea is simply to provide an option to exclude them from being masked.

Jean-Du, 2023-04-21:
This would function in a similar way to `area_thresholding` except that it's based on height. I recall @SylviaWhittle saying this is relatively straightforward to implement?

slackline, 2023-04-21:
Ah, ok so its not really a _removal_ then, rather its an aspect of masking and what to mask and what not to. :+1: 

I still think it would be useful to have some empirical data across a wide range of samples to do this. One image alone isn't a great basis for sampling and one unintended consequence might be that whilst it works on one image with obvious problems as linked that when applied to other images without such extremes some grains that are meant to be masked are not. 

As we go forward I'd like to design features correctly based on a range of samples as one of the drawbacks so far has been relying solely on `minicircles.spm` and then TopoStats being used on a range of different samples that throw up errors. If we have a good sample set to start with then we can design things better from the outset.