AFM-SPM/TopoStats #447: CSV file structure breaks with some images

SylviaWhittle, 2023-01-30:
**Describe the bug**
With some images, the file structure of the `folder_grainstats.csv` file breaks (incorrectly placed headers and commas). 

[image]

I think this might be something to do with detection of grains but then subsequent removal before grainstats outputs data on them? I apologise for not looking into this further, as I don't have the time currently. I will come back to this to add information.

**To Reproduce**

- Run topostats with any config on the following set of `.spm` files in the following folder. https://drive.google.com/drive/folders/1dflIB-qEGMahmdLoVXwDC_dm9Xksnv1h

Note that the `folder_grainstats.csv` file is incorrectly formatted.

- Config file

```
base_dir: ./by_sample/A4
output_dir: ./by_sample/A4
#base_dir: ./raphael_difficult_input/A3_MgCl2_008
#output_dir: ./raphael_difficult_output/A3_MgCl2_008_tilt_first
#base_dir: ./comparisons/slide_2_blue
#output_dir: ./comparisons/slide_2_blue
warnings: ignore
cores: 6
quiet: false
file_ext: .spm
loading:
  channel: Height
filter:
  run: true # Options : true, false
  threshold_method: std_dev # Options : otsu, std_dev, absolute
  otsu_threshold_multiplier: 1.0
  threshold_std_dev:
    lower: 10.0
    upper: 1.3
  threshold_absolute:
    lower: -1.0
    upper: 1.0
#  gaussian_size: 1.0121397464510862 # in px
  gaussian_size: 0.5
  gaussian_mode: nearest
  remove_scars:
    run: true
    removal_iterations: 2
    threshold_low: 0.250
    threshold_high: 0.666
    max_scar_width: 4
    min_scar_length: 10
grains:
  run: true # Options : true, false
  # Thresholding by height
  threshold_method: std_dev # Options : std_dev, otsu, absolute
  otsu_threshold_multiplier: 1.0
  threshold_std_dev:
    lower: 10.0
    upper: 1.3
  threshold_absolute:
    lower: -1.0
    upper: 1.0
  direction: upper # Options: upper, lower, both (defines whether to look for grains above or below thresholds or both)
  # Thresholding by area
  smallest_grain_size_nm2: 5000
  absolute_area_threshold:
    upper: [ 5000, 10000000 ] # above surface [Low, High] in nm^2 (also takes null)
    lower: [ null, null ] # below surface [Low, High] in nm^2 (also takes null)
grainstats:
  run: true # Options : true, false
  cropped_size: 500 # Length (in nm) of square cropped images (can take -1 for grain-sized box)
  save_cropped_grains: true # Options : true, false
dnatracing:
  run: true # Options : true, false
plotting:
  run: true # Options : true, false
  save_format: png # Options : see https://matplotlib.org/stable/api/_as_gen/matplotlib.pyplot.savefig.html
  image_set: core  # Options : all, core
  zrange: [-3, 4]  # low and high height range for core images (can take [null, null])
  colorbar: true  # Options : true, false
  axes: true # Options : true, false (due to off being a bool when parsed)
  cmap: nanoscope # Options : nanoscope, afmhot
  mask_cmap: blu
  histogram_log_axis: false # Options : true, false
  histogram_bins: 200 # Number of bins for histogram plots to use

```

**Expected behavior**
A properly formatted output `csv` file.

**(Not required if using Docker) Your computer configuration (please complete the following information):**
 - OS: MacOS
 - topostats version: `main` commit: [c9a6fd9](https://github.com/AFM-SPM/TopoStats/commit/c9a6fd99c311c468e3f5783db78e17d74a140af7)
 - Python version: 3.10

ns-rse, 2023-01-31:
I've started looking at this, part of the issue is there are some images without found grains.

Might be able to sort out the #403 at the same time, we'll see.

ns-rse, 2023-01-31:
Investigations.

## No Grains

If no grains are found the `utils.create_empty_dataframe()` function is called from `run_topostats.py`, this uses `ALL_STATISTICS_COLUMNS` for column names and was missing the `threshold` / `max_feret` / `min_feret` columns.

## `pd.concat()`

As each image is processed a Pandas DataFrame is generated holding the statistics, if no grains are found (or an error occurs) this is an empty dataframe so that something is returned and across all images the results can be concatenated into a single data frame using `pd.concat()` (without this `pd.concat()` complains about being unable to concatenate something of type `None`). When this is done though an additional column `Molecule Number` is added. This is because the above mentioned `create_empty_dataframe()` creates this as a column and doesn't set it as the index as is the case with the actual grain statistics.

Correcting this removes the extraneous empty column that was cropping up and including `Molecule Number` (the index) in the output means there are valid molecule numbers and address #403 (although `img_name` and `threshold` needs to be considered as these are not unique, perhaps a case for setting a Multi-Index so that they are unique?

I've corrected this and moved `create_empty_dataframe()` to `io` as it is more closely related to that than the catch all `utils`. I'll get a PR out this afternoon but any thoughts/comments on setting a Multi-Index of `["Image Name", "threshold", "Molecule Number"]` would be appreciated.

commit 38871c54 (2023-02-02): Fixes #447 extra column in CSV

Closes #447

* Fixes additional column in CSV output when images have no grains detected. This was caused by the empty dummy data
  frame not having the index set to `molecule_number` (nee `Molecule Number`) and so when concatenated via `pd.concat()`
  an additional column was introduced.
* Moves `utils.folder_grainstats()` to `io.folder_grainstats()`.
* Consistent case/format for variable names in CSV (all lower snake-case).
* CSVs now include `molecule_number` Closes #403

This will fail tests because it impacts regression tests. I've recent switched to using  [... cut]
commit e8d676f8 (2023-02-02): Merge pull request #449 from AFM-SPM/ns-rse/447-csv

Fixes #447 extra column in CSV
closed 2023-02-02 by ns-rse