AFM-SPM/TopoStats #470: Flag or skip images that are too small

SylviaWhittle, 2023-02-16:
**Is your feature request related to a problem? Please describe.**
Some images are just too small for the flattening algorithms to handle. This can cause errors to be thrown up in otherwise perfectly functional code, as with #381, which failed to fit a numpy polyfit to an image that was two pixels tall, as we would expect.

**Describe the solution you'd like**
Add a step to check if the image is smaller than say, 10x10, and if so, skip that image with a note in the console and mark it as failed to process.

ns-rse, 2023-03-04:
Do we have, or could we generate, parameters for what constitutes "too small"?

SylviaWhittle, 2023-05-19:
Completed as of #545

closed 2023-05-19 by SylviaWhittle