AFM-SPM/TopoStats #487: Add upper threshold option for grain height

SylviaWhittle, 2023-03-01:
Currently, grains are detected as being more than `threshold` above or below the background. Sometimes however, we want to be able to prevent the masking of data that lie too far from the background. For example, 'blobs' are often a lot higher than DNA, and we don't want them detected as parts of grains. 

Here is an attempt at drawing (side on) what this would look like for DNA data.

                                       XX            blob height
-----------------------XX------  "upper" threshold for data above the background
      oooo            oo        XXX          DNA height
--ooooooo----ooo---XXXX----- "lower" threshold for data above the background
======================== Background level

  ~~~~~~~      ~~~                         Data Mask

derollins, 2023-09-14:
Any update on this? I would like to implement this in my ["haribo rings"](https://github.com/AFM-SPM/TopoStats/tree/derollins/hariborings) project.

SylviaWhittle, 2023-09-14:
I'll do that now 👍

derollins, 2023-09-14:
To clarify, I'd like to just unmask parts of a grain that are above a threshold, not get rid of the whole grain.

SylviaWhittle, 2023-09-14:
@derollins Yep!