AFM-SPM/TopoStats #488: Z coordinates for molecule traces

SylviaWhittle, 2023-03-02:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

ns-rse, 2023-03-04:
JSON might be more succinct, will tag this onto #183 as it should be done under the refactored framework rather than the current module.

ns-rse, 2023-11-10:
#595 is a duplicate.

CSV is probably a better solution as it would be easier for more people to work with.

PR #790 (2024-02-12): Save grain trace data to HDF5 format
Closes #488 

This PR replaces #695. Instead of saving the trace data to JSON, we have decided to save it to the `.topostats` file in HDF5 format.

Experimentalists have requested that the grain trace data be accessible and readable in plain text format if possible, but we will add functions to make it easier to work with later on, adding a JSON dump function if absolutely necessary.
closed 2024-02-12 by SylviaWhittle