AFM-SPM/TopoStats #494: Publish to conda-forge?

stavrina, 2023-03-08:
Noticed that installing via anaconda requires a pip install.

Would it be possible to publish the package to conda-forge, instead of requiring a pip install within a conda environment?

pip installing within conda environments requires installing pip within that environment which adds another package dependency to the installation and update process

ns-rse, 2023-03-10:
Thanks for the suggestion @stavrina 

Currently we automate our releases to PyPI using GitHub Actions when a commit is tagged with a valid version (the workflow for that is [here](https://github.com/AFM-SPM/TopoStats/blob/main/.github/workflows/pypi.yaml)).

I've not yet built a package for [conda-forge](https://conda-forge.org/) but we would want to automate the process to align with releases to PyPI. A cursory search suggests there is an equivalent [GitHub Action : publish-conda-forge](https://github.com/marketplace/actions/publish-conda-forge) that we can likely leverage to achieve this.

We've still got a lot of refactoring and new features to add in the backlog but we are due to review the milestones and I'll make sure to mention this. If it makes life easier to `conda install topostats` then that is a good thing.

ns-rse, 2023-03-11:
> pip installing within conda environments requires installing pip within that environment which adds another package dependency to the installation and update process

By the way, I've just checked with a fresh install of Miniconda and when you create a new environment `pip` is automatically installed...

```bash
conda create --name test python=3.10
Collecting package metadata (current_repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /home/neil/miniconda3/envs/test

  added / updated specs:
    - python=3.10

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    openssl-1.1.1t             |       h7f8727e_0         3.7 MB
    pip-23.0.1                 |  py310h06a4308_0         2.6 MB
    python-3.10.9              |       h7a1cb2a_2        26.8 MB
    wheel-0.38.4               |  py310h06a4308_0          64 KB
    ------------------------------------------------------------
                                           Total:        33.2 MB

The following NEW packages will be INSTALLED:

  _libgcc_mutex      pkgs/main/linux-64::_libgcc_mutex-0.1-main 
  _openmp_mutex      pkgs/main/linux-64::_openmp_mutex-5.1-1_gnu 
  bzip2              pkgs/main/linux-64::bzip2-1.0.8-h7b6447c_0 
  ca-certificates    pkgs/main/linux-64::ca-certificates-2023.01.10-h06a4308_0 
  certifi            pkgs/main/linux-64::certifi-2022.12.7-py310h06a4308_0 
  ld_impl_linux-64   pkgs/main/linux-64::ld_impl_linux-64-2.38-h1181459_1 
  libffi             pkgs/main/linux-64::libffi-3.4.2-h6a678d5_6 
  libgcc-ng          pkgs/main/linux-64::libgcc-ng-11.2.0-h1234567_1 
  libgomp            pkgs/main/linux-64::libgomp-11.2.0-h1234567_1 
  libstdcxx-ng       pkgs/main/linux-64::libstdcxx-ng-11.2.0-h1234567_1 
  libuuid            pkgs/main/linux-64::libuuid-1.41.5-h5eee18b_0 
  ncurses            pkgs/main/linux-64::ncurses-6.4-h6a678d5_0 
  openssl            pkgs/main/linux-64::openssl-1.1.1t-h7f8727e_0 
  pip                pkgs/main/linux-64::pip-23.0.1-py310h06a4308_0  ### <<< pip is installed
  python             pkgs/main/linux-64::python-3.10.9-h7a1cb2a_2 
  readline           pkgs/main/linux-64::readline-8.2-h5eee18b_0 
  setuptools         pkgs/main/linux-64::setuptools-65.6.3-py310h06a4308_0 
  sqlite             pkgs/main/linux-64::sqlite-3.40.1-h5082296_0 
  tk                 pkgs/main/linux-64::tk-8.6.12-h1ccaba5_0 
  tzdata             pkgs/main/noarch::tzdata-2022g-h04d1e81_0 
  wheel              pkgs/main/linux-64::wheel-0.38.4-py310h06a4308_0 
  xz                 pkgs/main/linux-64::xz-5.2.10-h5eee18b_1 
  zlib               pkgs/main/linux-64::zlib-1.2.13-h5eee18b_0 

Proceed ([y]/n)? y

Downloading and Extracting Packages
                                                                                                                                                                                                                                                                                                                                
Preparing transaction: done                                                                                                                                                                                                                                                                                                     
Verifying transaction: done                                                                                                                                                                                                                                                                                                     
Executing transaction: done                                                                                                                                                                                                                                                                                                     
#
# To activate this environment, use
#
#     $ conda activate test
#
# To deactivate an active environment, use
#
#     $ conda deactivate
```

Thus there should be no need for an extra step of installing `pip` in Conda environments. Perhaps if you are having to manually install `pip` you might be able to `conda upgrade` to get this feature.

Getting an automated build pushed to conda-forge would still be useful, but I hope the above helps.

ns-rse, 2023-11-01:
Referencing this here should we get round to this.

[Grayskull](https://github.com/conda/grayskull) is a package that...

> can generate recipes for Python packages available on PyPI and also those not published on PyPI but available as GitHub repositories. Grayskull can also generate recipes for R packages published on CRAN. Future versions of Grayskull will support recipe generation for packages of other repositories such as Conan and CPAN etc..

Which might make this process easier. Probably the most challenging thing will be what I have encountered whilst trying to get TopoStats running in JupyterLike (see [jupyterlite-topostats](https://github.com/AFM-SPM/jupyterlite-topostats)) where many of our dependencies are not available in the repository for easy installation (or they were outdated/they recently changed their build system).

ns-rse, 2024-03-06:
Useful article if/when we get round to doing this

[Publish your Python package that is on PyPI to Conda Forge](https://www.pyopensci.org/python-package-guide/tutorials/publish-conda-forge.html).