AFM-SPM/TopoStats #507: Not all .spm files generate a processed image in specified file

LHolmes2, 2023-03-17:
## Checklist

Please try and tick off each of these items when filing the bug report. There are further instructions on each below.

* [ x] Describe the bug.
* [ x] Include the configuration file.
* [ x] Copy of the output.
* [ x] The exact command that failed. This is what you typed at the command line, including any options.
* [x ] TopoStats version, this is reported by `run_topostats --version`
* [ x] Operating System and Python Version

## Describe the bug
When running TopoStats using `git`, `pip install -e .` , not all .spm files generate a processed image. 

## Copy of the output

```
(topostats-git) libbyholmes@dyn221187 TopoStats % run_topostats --config_file dummy_config.yaml
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] The YAML configuration file is valid.
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] The YAML plotting configuration file is valid.
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Configuration run options are consistent, processing can proceed.
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Configuration file loaded from      : dummy_config.yaml
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Scanning for images in              : /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Output directory                    : topostats/images/15032023DNA_cleaning
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Looking for images with extension   : .spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Images with extension .spm in /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images : 7
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Thresholding method (Filtering)     : std_dev
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Thresholding method (Grains)        : std_dev
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Extracting image from /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230315_piAN7.0_00004.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Loading image from : /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230315_piAN7.0_00004.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] [20230315_piAN7.0_00004] : Loaded image from : /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230315_piAN7.0_00004.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] [20230315_piAN7.0_00004] : Extracted channel Height
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] [20230315_piAN7.0_00004] : Pixel to nm scaling : 1.953125
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Extracting image from /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230306_L22_2B_3ng_pian7_0_00000.0_00019.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Loading image from : /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230306_L22_2B_3ng_pian7_0_00000.0_00019.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] [20230306_L22_2B_3ng_pian7_0_00000.0_00019] : Loaded image from : /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230306_L22_2B_3ng_pian7_0_00000.0_00019.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] [20230306_L22_2B_3ng_pian7_0_00000.0_00019] : Extracted channel Height
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] [20230306_L22_2B_3ng_pian7_0_00000.0_00019] : Pixel to nm scaling : 1.953125
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Extracting image from /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230309_piAN7_gelexcision2_NiCl2_3mM_HEPES_20mM.0_00003.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Loading image from : /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230309_piAN7_gelexcision2_NiCl2_3mM_HEPES_20mM.0_00003.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] [20230309_piAN7_gelexcision2_NiCl2_3mM_HEPES_20mM.0_00003] : Loaded image from : /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230309_piAN7_gelexcision2_NiCl2_3mM_HEPES_20mM.0_00003.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] [20230309_piAN7_gelexcision2_NiCl2_3mM_HEPES_20mM.0_00003] : Extracted channel Height
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] [20230309_piAN7_gelexcision2_NiCl2_3mM_HEPES_20mM.0_00003] : Pixel to nm scaling : 1.953125
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Extracting image from /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230309_piAN7_gelexcision1_NiCl2_3mM_HEPES_20mM.0_00002.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Loading image from : /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230309_piAN7_gelexcision1_NiCl2_3mM_HEPES_20mM.0_00002.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] [20230309_piAN7_gelexcision1_NiCl2_3mM_HEPES_20mM.0_00002] : Loaded image from : /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230309_piAN7_gelexcision1_NiCl2_3mM_HEPES_20mM.0_00002.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] [20230309_piAN7_gelexcision1_NiCl2_3mM_HEPES_20mM.0_00002] : Extracted channel Height
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] [20230309_piAN7_gelexcision1_NiCl2_3mM_HEPES_20mM.0_00002] : Pixel to nm scaling : 1.953125
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Extracting image from /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230306_L22_2A_piAN7_3ng0_00000.0_00012.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Loading image from : /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230306_L22_2A_piAN7_3ng0_00000.0_00012.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] [20230306_L22_2A_piAN7_3ng0_00000.0_00012] : Loaded image from : /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230306_L22_2A_piAN7_3ng0_00000.0_00012.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] [20230306_L22_2A_piAN7_3ng0_00000.0_00012] : Extracted channel Height
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] [20230306_L22_2A_piAN7_3ng0_00000.0_00012] : Pixel to nm scaling : 1.953125
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Extracting image from /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230306_L22_2A_piAN7_3ng0_00000.0_00013.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Loading image from : /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230306_L22_2A_piAN7_3ng0_00000.0_00013.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] [20230306_L22_2A_piAN7_3ng0_00000.0_00013] : Loaded image from : /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230306_L22_2A_piAN7_3ng0_00000.0_00013.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] [20230306_L22_2A_piAN7_3ng0_00000.0_00013] : Extracted channel Height
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] [20230306_L22_2A_piAN7_3ng0_00000.0_00013] : Pixel to nm scaling : 1.953125
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Extracting image from /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230306_L22_2A_piAN7_3ng0_00000.0_00011.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] Loading image from : /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230306_L22_2A_piAN7_3ng0_00000.0_00011.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] [20230306_L22_2A_piAN7_3ng0_00000.0_00011] : Loaded image from : /Users/libbyholmes/Documents/PhD/topostats-tutorial/TopoStats/topostats/images/15032023DNA_cleaning/20230306_L22_2A_piAN7_3ng0_00000.0_00011.spm
[Fri, 17 Mar 2023 13:49:57] [INFO    ] [topostats] [20230306_L22_2A_piAN7_3ng0_00000.0_00011] : Extracted channel Height
[... cut ...]
[... 406 lines cut ...]
packaging==23.0
pandas==1.5.3
Pillow==9.4.0
pyfiglet==0.8.post1
pyparsing==3.0.9
pySPM==0.2.23
python-dateutil==2.8.2
pytz==2022.7.1
PyWavelets==1.4.1
PyYAML==6.0
ruamel.yaml==0.17.21
ruamel.yaml.clib==0.2.7
schema==0.7.5
scikit-image==0.20.0
scikit-learn==1.2.2
scipy==1.10.1
seaborn==0.12.2
six==1.16.0
threadpoolctl==3.1.0
tifffile==2023.3.15
-e git+https://github.com/AFM-SPM/TopoStats.git@2c0d43b944a94ead1b530a2473f04cab9360c662#egg=topostats
tqdm==4.65.0
```

## Additional context

ns-rse, 2023-03-21:
Thanks for reporting this @LHolmes2 really useful.

One thing which I've omitted and will update shortly in the template is to help identify which image causes the error. This can be achieved by reducing the `cores: 1` so that the images are not processed in parallel and are instead done so sequentially. That way we can identify which image is causing the problem and investigate.

If you've a moment to re-run the analysis in TopoStats with this option and report back what image (and if possible share it, link to GoogleDrive is ok) that would be really useful.

**NB** You should also be able to modify the number of cores used by using the `-cores 1` flag  (or `-j 1`) on the command line in conjunction with your config file (as command line options over-ride those in the configuration file).

```bash
run_topostats --config_file dummy_config.yaml --cores 1
```

LHolmes2, 2023-03-22:
Hi Neil,

Thanks very much for this, I'll try re-run this some time this week for you
and report back.

Thanks,
Libby

closed 2023-05-09 by ns-rse