AFM-SPM/TopoStats #508: Speed up tests

ns-rse, 2023-03-18:
Tests take a long time. There are two areas that this can be done (they should be treated as separate tasks/issues, capturing here for now).

* [ ] Trim down the array from `minicircle.spm` to a smaller array to speed up all the processing steps.
* [ ] We can leverage [Python caching in GitHub Actions](https://github.blog/changelog/2021-11-23-github-actions-setup-python-now-supports-dependency-caching/) to speed up installation of environments.

SylviaWhittle, 2023-04-28:
I think that the issue with speed is mostly my fault. The script `remove_scars()` in `scars.py` gets run 4 times by default when processing images (twice during initial flattening and twice during proper flattening). Based on some brief experiments, it seems that scar removal makes up about 70% of the time cost of topostats (based off of a run of minicircle.spm on M1 Pro Apple Silicon processor, using the default config, 25 seconds runtime, 16 seconds of which are spent on scar removal). 

I seriously doubt that I would be able to optimise the algorithm, so perhaps we could discuss feasibility of offloading the scar removal process to a faster language (C, Fortran, Rust, etc, or perhaps incorporating systems like Cython). This could cause issues in keeping the software platform agnostic however...

Do you have any thoughts on this @ns-rse ?

ns-rse, 2023-05-03:
Thanks for the insight and thoughts on this @SylviaWhittle .

I think that whilst performance of `remove_scars()` impacts the speed of the test suite overall I would see that as a separate issue to be tackled independently of speeding up the tests in general.

My thinking with this issue is that processing time could be vastly reduced if we had a smaller scan to process as `minicircle.spm` is quite large. Also with regards to `remove_scars()` aside from the regression tests there may be steps where we can improve the unit tests and test say grain/grainstats/tracing functionality without having to enable scar removal.

SylviaWhittle, 2023-05-04:
Hey @ns-rse , @derollins has this image which is 256x256 and processes in 6 seconds on my machine versus 26 seconds (a 77% improvement) for minicircle.spm.

[image]

[image]

The image has both linear and circular molecules, and has about 32 grains of varying sizes, so plenty to test with.

What do you think?

PR #633 (2023-07-28): Speed up tests
Closes #508 

This PR is aimed at increasing the speed of tests. It does this by replacing the `(1024, 1024)` `minicircle.spm` testing image, with a slice of this image, namely `minicircle[-296:,-296:]` and compressed down from `(256, 256)` to `(64, 64)`. This image is saved in the ✨`.topostats`✨  file format instead of `.spm`.

I have checked that it flattens appropriately, grains are detected, and traces are produced despite this low resolution. (It also looks really cute).
[image]
[image]
[image]

## Optimisation

Previously, running the automated test suite on GitHub would t [... cut]
closed 2023-07-28 by SylviaWhittle