AFM-SPM/TopoStats #535: Error when `colorbar` is turned off

derollins, 2023-04-20:
## Checklist

Please try and tick off each of these items when filing the bug report. There are further instructions on each below.

* [x] Re-run analysis with `run_topostats --core 1`.
* [x] Describe the bug.
* [x] Include the configuration file.
* [x] Copy of the output.
* [x] The exact command that failed. This is what you typed at the command line, including any options.
* [ ] TopoStats version, this is reported by `run_topostats --version`
* [ ] Operating System and Python Version

## Describe the bug
When the `colorbar` option is changed to false Topostats 

## Copy of the output

Please copy and paste the output that is shown below within the `\`` (triple-backticks).

```
[Thu, 20 Apr 2023 13:38:06] [INFO    ] [topostats] [13-gaussian_filtered] : Image saved to : C:\Users\Work\OneDrive\Documents\Uni\Research\Data\My_Data\20230412_Cas9_minicircles\output_noscale\processed\20230412_20230412_Cas9_RNA_complex_3ng_TRIS_10mM.0_00014\filters\13-gaussian_filtered.png
Processing images from C:\Users\Work\OneDrive\Documents\Uni\Research\Data\My_Data\20230412_Cas9_minicircles, results ar
multiprocessing.pool.RemoteTraceback:
"""
Traceback (most recent call last):
  File "C:\Users\Work\anaconda3\envs\topostats_git\lib\multiprocessing\pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "C:\Users\Work\OneDrive\Documents\Uni\Research\Code\Topostats_git\topostats\processing.py", line 124, in process_scan
    ).plot_and_save()
  File "C:\Users\Work\OneDrive\Documents\Uni\Research\Code\Topostats_git\topostats\plottingfuncs.py", line 176, in plot_and_save
    self.save_array_figure()
  File "C:\Users\Work\OneDrive\Documents\Uni\Research\Code\Topostats_git\topostats\plottingfuncs.py", line 257, in save_array_figure
    plt.imsave(
  File "C:\Users\Work\anaconda3\envs\topostats_git\lib\site-packages\matplotlib\pyplot.py", line 2136, in imsave
    return matplotlib.image.imsave(fname, arr, **kwargs)
  File "C:\Users\Work\anaconda3\envs\topostats_git\lib\site-packages\matplotlib\image.py", line 1630, in imsave
    rgba = sm.to_rgba(arr, bytes=True)
  File "C:\Users\Work\anaconda3\envs\topostats_git\lib\site-packages\matplotlib\cm.py", line 511, in to_rgba
    x = self.norm(x)
  File "C:\Users\Work\anaconda3\envs\topostats_git\lib\site-packages\matplotlib\colors.py", line 1345, in __call__
    raise ValueError("minvalue must be less than or equal to maxvalue")
ValueError: minvalue must be less than or equal to maxvalue
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\Work\anaconda3\envs\topostats_git\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Work\anaconda3\envs\topostats_git\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\Work\anaconda3\envs\topostats_git\Scripts\run_topostats.exe\__main__.py", line 7, in <module>
  File "C:\Users\Work\OneDrive\Documents\Uni\Research\Code\Topostats_git\topostats\run_topostats.py", line 235, in main
    for img, result in pool.imap_unordered(
  File "C:\Users\Work\anaconda3\envs\topostats_git\lib\multiprocessing\pool.py", line 873, in next
    raise value
ValueError: minvalue must be less than or equal to maxvalue
```

## Include the configuration file.

If no configuration file was specified with the `-c`/`--config-file` option the defaults were used, please use the
`run_topostats --create-config-file crash.yaml` to save these to the `crash.yaml` file and copy the contents below.

``` yaml
# Sample configuration file auto-generated : 2023-04-17 16:43:55
base_dir: C:\Users\Work\OneDrive\Documents\Uni\Research\Data\My_Data\20230412_Cas9_minicircles
output_dir: C:\Users\Work\OneDrive\Documents\Uni\Research\Data\My_Data\20230412_Cas9_minicircles\output_noscale
log_level: info
cores: 1
[... 221 lines cut ...]
soupsieve @ file:///C:/b/abs_fasraqxhlv/croot/soupsieve_1666296394662/work
stack-data @ file:///opt/conda/conda-bld/stack_data_1646927590127/work
terminado @ file:///C:/b/abs_25nakickad/croot/terminado_1671751845491/work
threadpoolctl==3.1.0
tifffile==2023.2.3
tinycss2 @ file:///C:/b/abs_52w5vfuaax/croot/tinycss2_1668168823131/work
toml @ file:///tmp/build/80754af9/toml_1616166611790/work
tomli @ file:///C:/Windows/TEMP/abs_ac109f85-a7b3-4b4d-bcfd-52622eceddf0hy332ojo/croots/recipe/tomli_1657175513137/work
-e git+https://github.com/AFM-SPM/TopoStats.git@bd06b70731613684f400d2e9d5fbd8d78295cf4a#egg=topostats
tornado @ file:///C:/ci/tornado_1662476985533/work
tqdm==4.64.1
traitlets @ file:///C:/b/abs_e5m_xjjl94/croot/traitlets_1671143896266/work
typing_extensions @ file:///C:/b/abs_89eui86zuq/croot/typing_extensions_1669923792806/work
urllib3 @ file:///C:/b/abs_9bcwxczrvm/croot/urllib3_1673575521331/work
wcwidth @ file:///Users/ktietz/demo/mc3/conda-bld/wcwidth_1629357192024/work
webencodings==0.5.1
websocket-client @ file:///C:/ci_310/websocket-client_1642093970919/work
widgetsnbextension @ file:///C:/ci/widgetsnbextension_1645009839917/work
win-inet-pton @ file:///C:/ci_310/win_inet_pton_1642658466512/work
wincertstore==0.2
```

## Additional context

Add any other context about the problem here.

SylviaWhittle, 2023-04-25:
@derollins Sorry I may be misunderstanding or being a bit dim, but I can't seem to reproduce this bug by setting `colorbar` to `false`. 

[image]

It successfully plots the image without a colour bar:

[image]

I can only get it to throw the error listed in the output by swapping the `zrange: [-3, 4]` to `zrange: [4, -3]`:

[image]

[image]

And this is expected to crash since 4 cannot be the minimum if -3 is the maximum value.

ns-rse, 2023-05-09:
It looks to me like this was an error with the configuration as the plotting configuration contains not only `colorbar: false` but the `zrange:` is invalid as @SylviaWhittle notes (see also entry in table of [configuration](https://afm-spm.github.io/TopoStats/main/configuration.html)).

```yaml
plotting:
  run: true
  save_format: png
  image_set: core
  zrange:
  - 4
  - 0
  colorbar: false
  axes: false
  cmap: afmhot
  mask_cmap: blu
  histogram_log_axis: false
  histogram_bins: 200
  dpi: 100
```

#554 clarifies this in the documentation and `topostas/default_config.yaml`.

commit f7331b81 (2023-05-09): Clarifying documentation for zrange

Closes #535

This wasn't an error in the code but hopefully clarifying the documentation may help avoid the problem in the future.
PR #554 (2023-05-09): Clarifying documentation for zrange
Closes #535

This wasn't an error in the code but hopefully clarifying the documentation may help avoid the problem in the future.
closed 2023-05-09 by ns-rse