AFM-SPM/TopoStats #601: Further Modularisation

SylviaWhittle, 2023-06-12:
**Is your feature request related to a problem? Please describe.**
TopoStats has been made more modular with the merging of recent PRs #540 and #600 (thanks @ns-rse). It would be good to have further work enabling the independent function of the different steps.

Here's my initial rough idea on how this might look...

```
[entry point] ----- FLATTENING in: [raw data, pixel to nanometre scaling] out: [flattened image array, optional plots?]

[entry point] ----- GRAIN FINDING & GRAINSTATS in: [flattened image, pixel to nanometre scaling] out: [grain masks, grain stats file, optional plots?]

[entry point] ------ DNATRACING in: [grain masks, pixel to nanometre scaling, grain stats file] out: [dna traces, images, etc, dna stats file, optional plots?]

[entry point] ------ PLOTTING (summary statistics) in [ stats files ] out: [distribution plots and visualisations]

```

This could enable easier feature additions and perhaps easier testing procedures.

derollins, 2023-06-13:
From a heavy user perspective this would be very useful. For a complex dataset I usually have to iterate the analysis at each step (processing, image plotting, masking etc.) and can have different parameters for each part for each part of the analysis. This currently can be very time consuming especially when I sometimes need to use the 'all' image set and also topostats is filling up my laptop storage very quickly since I currently just change the name of the output file for each run I want to compare so get everything multiple times.

It would be great if once the raw data had been flattened I could simply use that flattened data to make new images (different scales, colours, formats etc.) and optimise masking from the flattened image rather than having to completely reprocess everything.

ns-rse, 2023-09-21:
Feels like we're well on the way to achieving this with the refactoring @SylviaWhittle undertook which was merged in #613.

What more needs doing to round this out? Is it better documentation, tutorials on how to use the new command line interface (CLI), are we missing modules for anything, perhaps tracing/skeletonisation aspects which are blocked by the refactoring that is a work in (slow) progress.