AFM-SPM/TopoStats #610: Prune skeletons

ns-rse, 2023-06-26:
#600 and #609 restructured `dnatracing.traceDna` to work on a single grain and introduced the ability to select skeletonisation using [scikit-image](https://scikit-image.org) methods ([Zhang, Lee or Thinning](https://scikit-image.org/docs/stable/auto_examples/edges/plot_skeleton.html#skeletonize)).

These typically return skeletons with lots of branches which need "pruning".

One candidate library for achieving this is [Fil-Finder](https://fil-finder.readthedocs.io/en/latest/tutorial.html#skeletons-pruning-length) and this takes an interesting approach to pruning which incorporates determining the length of objects at the same time, from the [docs]()

> The first analysis step includes two parts: pruning the skeleton structures and finding the filament lengths. The first part removes small/unimportant spurs on the skeletons. To ensure important portions of the skeleton are retained, however, both parts are performed together.
> 
> Each skeleton is converted into a graph object using the [networkx](https://networkx.github.io/) package. We use the graph to find the longest path through the skeleton, which is used to define the structure’s length. All branches in the skeleton away from this longest path are eligible to be pruned off.

Reading further though this might present yet another alternative to the height based approach which has been investigated by @MaxGamill-Sheffield as it goes on to state...

> The default settings use minimum skeleton and branch lengths based off of the beam size. To be kept, a branch must be at least three times the length of the beam and a skeleton must have a length of 5 times the beam. Practically, this will only remove very small features.
> 
> These parameters, and ones related to the pruning, can be manually set:
> 
> * `prune_criteria` – The criteria for removing a branch can be altered. The default (‘all’) uses a mix of the average intensity along the branch and its length. The length alone can be used for pruning with `prune_criteria='length'`. All branches below this length will be removed. Finally, only the intensity can be used for pruning (`prune_criteria='intensity'`). A branch is kept in this case by comparing the average intensity along the branch to the average over the whole filament. The critical fraction that determines whether a branch is important is set by `relintens_thresh`.
> * `relintens_thresh` – Set the critical intensity comparison for intensity-based pruning.

Whilst designed to work with astropy and clouds of brightness the images we are working with here seem quite conducive to these methods since our "intensity" is the height of the molecule.

At a bare minimum we should investigate this library's utility for pruning, however it may provide a useful method for skeletonisation and calculation of `end_to_end_distance` too.

ns-rse, 2023-06-29:
Some digging around led me to the [skan](https://github.com/jni/skan/) package which was originally designed for working with AFM images.

There is an undocumented [`Skeleton.prune_paths()`](https://github.com/jni/skan/issues/118) method and work is in progress to extend this to iterratively prune skeletons.

ns-rse, 2023-07-31:
As this involves work on `skan` I've been documenting progress in [Skan issue 206](https://github.com/jni/skan/issues/206).

SylviaWhittle, 2024-11-13:
Max added good, configurable pruning in #932