AFM-SPM/TopoStats #615: Add support for .TIFF files

SylviaWhittle, 2023-07-06:
We have had a request for .TIFF file format support. Initially we thought that this is nonsensical but apparently `.tiff` files can have metadata stored inside them, so we can access the pixel to nanometre scaling factor.

I'll update this issue when I find out more info

derollins, 2023-07-13:
Just a thought on this, when opening TIFF files in other microscopy image analysis programmes then you can input a scaling factor manually. Would this is doable in topostats?

SylviaWhittle, 2023-07-13:
@derollins It would be possible, though there would be much discussion and argument about how exactly this should be done.

One option could be to have a configuration parameter in topostats...

```yaml
pixel_to_nm_scaling_factor: null # Manually specify the pixel to nanometre scaling factor. Will apply to ALL images that are processed, so all images MUST be the same resolution.
```