AFM-SPM/TopoStats #617: Method of Quantifying Protein Location on DNA Strand

neumagen, 2023-07-10:
Hello! 

I am new to TopoStats, and am using it as an alternative to an outdated method, which relied on an abundance of manual input. 

The goal of the research project is incubating certain proteins with different types of DNA to see the potential localization of said protein on the strand. This would give an indication as to the functionality of the protein. 

Currently, we visually look and make a qualitative observation as to whether the protein is interacting with the DNA strand, and if so, whether it is at the end or in the middle of the DNA.

Is it possible to take the XY coordinates of the proteins and correlate them to the XY coordinates of the DNA? I know I'm oversimplifying this!

Thanks

SylviaWhittle, 2023-07-31:
This seems like it should be do-able. If I'm understanding this correctly, we could mask again with a higher threshold to identify proteins, then try to identify proteins touching DNA using the masks and distances, then take the splined trace coordinates of any attached DNA to each protein, courtesy of @MaxGamill-Sheffield 's work and find the point of the DNA spline that is closest to the centre of the protein blob, hence telling us where along the DNA the protein is bound?

neumagen, 2023-07-31:
Yes that sounds exactly like it @SylviaWhittle!

I know of 4 or 5 different projects from various levels of investigators (MSc, PhD, and research associate) in different labs that are only interested in quantifying protein localization (which has never been done before using this technology). I have told them that I could help set them up with TopoStats for analysis if it comes to fruition. We also have a massive multi-faculty imaging facility here at McMaster with several AFM machines where I would love to set up TopoStats in their data analysis suite if this works out!
 
Not trying to pressure anyone because I know it’s a lot of work, but just trying to prove that there is a demand for this type of analysis!

SylviaWhittle, 2023-08-01:
@neumagen 
This is now being worked on. A branch has been created, `SylviaWhittle/617-protein-bind-site` and I'll write about progress on here.

neumagen, 2023-08-02:
@SylviaWhittle This is amazing!! Exactly what I wanted. 

For the intended data, I think it would be useful to have a couple different measuring methods. Is the protein at the end of the DNA strand (no = 0, yes = 1)? If the answer is no, then provide the nm measurement along the DNA strand. 

Not sure about the circular molecules, maybe someone else has an idea? My group has lab meeting in an hour and I will pick their brains about what information they want from this :) 

For the branching molecules, could we identify if the protein is at the intersection of the branch? Maybe assign 0 as not being at the intersection and 1 as being at the intersection? If yes, where (in nm) the branch occurs.

This way we could get a statements like "DNA grain 1 has 2 proteins bound at either end of the strand and 3 further ones at 12, 35, 41 nm along the backbone" and "DNA grain 2 is branching and has 1 protein at its intersection at 10 nm, no proteins are bound at either end of the strand, but there are 2 further ones at 35 and 41 nm along the backbone"

neumagen, 2023-08-02:
@SylviaWhittle 

Okay I talked to my lab and here is what they had to say: 

For the circular DNA, they are only interested in the total protein count on that backbone. This is so we can compare the total number of proteins on a strand with no DNA damage vs total number of proteins on DNA with double strand breaks. 

They would also like a histogram of protein count vs DNA length, i.e. 3 proteins are found at one end, 2 at 12 nm, 1 at 24 nm, 2 at the other end.  DNA length might have to be a percentage or ratio rather than actual nm  because of the varying backbone lengths. 

Finally, they want to know if there is a way to quantify these particular proteins at DNA strand "bubbles", such as those found with gyrase activity. I am working on obtaining some clear sample images from a couple different people.