AFM-SPM/TopoStats #630: Cats branch intermittent thread crash & hang

MaxGamill-Sheffield, 2023-07-20:
## Checklist

Please try and tick off each of these items when filing the bug report. There are further instructions on each below.

* [ /] Re-run analysis with `run_topostats --core 1`.
* [ /] Describe the bug.
* [ ] Include the configuration file.
* [ ] Copy of the output.
* [ ] The exact command that failed. This is what you typed at the command line, including any options.
* [ ] TopoStats version, this is reported by `run_topostats --version`
* [ ] Operating System and Python Version

## Describe the bug
On the cats branch, running topostats can sometimes cause a thread to hang when processing an image. It seems to hang after all the images are created (identified by a print statement just before the `return` in `processing.py`) possibly suggesting that it is the rejoining of the thread.

## Copy of the output
See the terminal saved output:
[Terminal Saved Output.txt](https://github.com/AFM-SPM/TopoStats/files/12094607/Terminal.Saved.Output.txt)

See the crash report:
[crash report.txt](https://github.com/AFM-SPM/TopoStats/files/12094609/crash.report.txt)

## Include the configuration file

``` yaml
# Config file generated 2023-07-07 13:25:12
# For more information on configuration and how to use it:
# https://afm-spm.github.io/TopoStats/main/configuration.html
base_dir: ./bigtest/cats/Curated cats #split cats/13 # Curated cats # ./test2/cat # Directory in which to search for data files
output_dir: ./bigtest/cats/Curated cats #split cats/13 # Directory to output results to
#base_dir: ./test2/cat # Directory in which to search for data files
#output_dir: ./test2/cat # Directory to output results to
log_level: info # Verbosity of output. Options: warning, error, info, debug
cores: 1 # Number of CPU cores to utilise for processing multiple files simultaneously.
file_ext: .spm # File extension of the data files.
loading:
  channel: Height # Channel to pull data from in the data files.
filter:
  run: true # Options : true, false
  row_alignment_quantile: 0.5 # below values may improve flattening of larger features
  threshold_method: std_dev # Options : otsu, std_dev, absolute
  otsu_threshold_multiplier: 1.0
  threshold_std_dev:
    below: 10.0 # Threshold for data below the image background
    above: 1.0 # Threshold for data above the image background
  threshold_absolute:
    below: -1.0 # Threshold for data below the image background
    above: 1.0 # Threshold for data above the image background
  gaussian_size: 1.0121397464510862 # Gaussian blur intensity in px
  gaussian_mode: nearest
  # Scar remvoal parameters. Be careful with editing these as making the algorithm too sensitive may
  # result in ruining legitimate data.
  remove_scars:
    run: true
    removal_iterations: 2 # Number of times to run scar removal.
    threshold_low: 0.250 # below values make scar removal more sensitive
    threshold_high: 0.666 # below values make scar removal more sensitive
    max_scar_width: 4 # Maximum thichness of scars in pixels.
    min_scar_length: 16 # Minimum length of scars in pixels.
grains:
  run: true # Options : true, false
  # Thresholding by height
  threshold_method: std_dev # Options : std_dev, otsu, absolute
  otsu_threshold_multiplier: 1.0
  threshold_std_dev:
    below: 10.0 # Threshold for grains below the image background
    above: 1.2 # Threshold for grains above the image background
  threshold_absolute:
    below: -1.0 # Threshold for grains below the image background
    above: 1.0 # Threshold for grains above the image background
  direction: above # Options: above, below, both (defines whether to look for grains above or below thresholds or both)
  # Thresholding by area
  smallest_grain_size_nm2: 50 # Size in nm^2 of tiny grains/blobs (noise) to remove, must be > 0.0
  absolute_area_threshold:
    above: [ 300, null ] # above surface [Low, High] in nm^2 (also takes null)
    below: [ null, null ] # below surface [Low, High] in nm^2 (also takes null)
grainstats:
  run: true # Options : true, false
  edge_detection_method: binary_erosion # Options: canny, binary erosion. Do not change this unless you are sure of what this will do.
  cropped_size: -1 # Length (in nm) of square cropped images (can take -1 for grain-sized box)
dnatracing:
  run: true # Options : true, false
  min_skeleton_size: 10 # Minimum number of pixels in a skeleton for it to be retained.
  skeletonisation_method: joe # Options : zhang | lee | thin | topostats
  pruning_method: joe # Options : joe (more to come)
  pad_width: 20 # Cells to pad grains by when tracing
#  cores: 1 # Number of cores to use for parallel processing
plotting:
  run: true # Options : true, false
  save_format: png # Options : see https://matplotlib.org/stable/api/_as_gen/matplotlib.pyplot.savefig.html
  pixel_interpolation: null # Options : https://matplotlib.org/stable/gallery/images_contours_and_fields/interpolation_methods.html
  image_set: all  # Options : all, core
  zrange: [null, null]  # low and high height range for core images (can take [null, null]). low <= high
  colorbar: true  # Options : true, false
  axes: true # Options : true, false (due to off being a bool when parsed)
  cmap: nanoscope # Options : nanoscope, afmhot, gwyddion
  mask_cmap: blu # Options : blu, jet_r and any in matplotlib
  histogram_log_axis: false # Options : true, false
  histogram_bins: 200 # Number of bins for histogram plots to use
summary_stats:
  run: true # Whether to make summary plots for output data
  config: null
```

## To Reproduce
Use files found here: https://drive.google.com/drive/folders/16zJbzBA8UrOpYR1jXHugCa2TSjWi4TQ7?usp=drive_link

And run topostats under the cats branch. This may require multiple runs to get it to hang.

## Output
Images only, none of the statistic files

## TopoStats version
``` bash
2.1.1.dev242+g5e038684.d20230707
```

## Operating System and Python Version

### Operating System

Please let us know what operating system you are using, if you have used more than one then tick all boxes.

* [ ] Windows
* [ ] MacOS Intel (pre-2021)
* [ /] MacOS M1/M2 (post-2021)
* [ ] GNU/Linux (please include distribution)

### Python Version

Please let us know the version of Python you are using, paste the results of `python --version`

``` bash
3.10.6
```

### Optional : Python Packages
Main one will probably be `topoly`

## Additional context

Add any other context about the problem here.

ns-rse, 2023-10-13:
Can this be closed now @MaxGamill-Sheffield I think I recall you saying you'd solved it.

MaxGamill-Sheffield, 2023-10-17:
Kind of, but it's not a permanent solution. Sylvia shared a link saying that Python may be removing the GIL which I believe is causing the error. As the fix below is temporary, should I still close the issue?:

**As a temporary fix**, when the program hangs at the end of running all the files, you can or kill (or "force quit" on mac) the hanging python process and a new process will spawn. This will finish running TopoStats and produce the final output statistics file :)

ns-rse, 2023-10-20:
It will be a quite a while until the Global Interpreter Lock (GIL) issue is removed from Python.

If there is nothing that can be done to work around perhaps writing an FAQ article for the documentation would be sensible explaining what the issue is and how to handle it if users encounter it would be reasonable. That way the knowledge is out there and shared with users to find (should they go looking, even if they ask its easier to paste a link to the solution than have to write the same solution again or dig an old email out and copy/paste).

MaxGamill-Sheffield, 2023-10-26:
Good idea, added to the Q and A section :)

closed 2023-10-26 by MaxGamill-Sheffield

ns-rse, 2023-10-27:
[Why does TopoStats hang once it look like its finished processing? · AFM-SPM/TopoStats · Discussion #710](https://github.com/AFM-SPM/TopoStats/discussions/710)

ns-rse, 2023-10-27:
Just a thought on this, but is this perhaps something to do with [Topoly](https://topoly.cent.uw.edu.pl/tutorial.html) and how it has been used/integrated?

Has anyone else seen this hang using TopoStats _without_ Topoly?

If not that is a strong indicator of where to look for the problem. Most of the parallelisation that is done so far is such that there shouldn't be any problems from the Global Interpreter Lock because each image is analysed on its own. If this isn't the case with how integration with Topoly has been done it would be the place to start. A crude guess would be that perhaps a class is instantiated and used to work out topology of two entangled molecules.

MaxGamill-Sheffield, 2023-10-27:
Yep it is Topoly, but I've not the time spare to understand specifically what it is or build a multiprocessing test case right now.

When I was testing Topoly using PD Codes (way back) it would often crash and I assumed it was taking up more memory than it should when trying to calculate a topology for a PD Code that wasn't perfect. This was slightly solved when moving to using the `Jones` algorithm instead of `homfly` but still on occasion would crash.

ns-rse, 2023-10-30:
We'll have to work on that then and work out the problem, probably worth seeking assistance upstream too.