AFM-SPM/TopoStats #645: Move DPI setting to plotting dictionary

ns-rse, 2023-08-04:
We only need high resolution images for traced plots and having it as a global setting means that plotting of scans slows down too much.

By moving the configuration of DPI to the plotting dictionary we will have finer grained control over the resolution of images.

ns-rse, 2023-09-11:
@MaxGamill-Sheffield @Jean-Du @SylviaWhittle :+1: 

Looking to work on this but wondering what DPI settings people have been using for traced plots?

The current global value which comes through is `config["plotting"]["dpi"] = 100` (set in `topostats/default_config.yaml`).

If moving to `topostats/plotting_dictionary.yaml` most will also have `dpi = 100` but...

1. What would a reasonable default be for the traced plots that are configured under `all_molecule_traces` and `single_molecule_trace`?
2. Are there any other plots beyond these two that would require a higher DPI?

MaxGamill-Sheffield, 2023-09-11:
Hey Neil, I've already got some of this in my branch but I'm not sure how easy it would be to move over.

In answer to your questions:
1. I use `dpi=800` to ensure the single pixels aren't interpolated over / ignored and this has worked pretty well for all my images. If the single molecule traces are the cropped ones then they might be ok to be lower as each pixel will be larger.
2. Depending on how you're drawing splines on the mpl image, possibly those too. Although I believe `all/single_molexule_traces` is actually the ordered trace and not the final "splined" trace - but my memory might need refreshing.

ns-rse, 2023-09-11:
Thanks @MaxGamill-Sheffield that's really useful, I'll have a look at your branch to guide me too.

commit 1027de31 (2023-09-13): Moves DPI configuration from default_config to plotting_dictionary

Closes #645

Moves specification of a global DPI for all images from `topostats/default_config.yaml` to per-plot in
`topostats/plotting_config.yaml` with default of `100` for most values but `800` for existing plots of traces.

Updates validation rules and documentation and I've taken the opportunity to add missing Description to two rows in the
documentation covering the `summary_config.yaml`.
PR #662 (2023-09-13): Moves DPI configuration from default_config to plotting_dictionary
Closes #645

Moves specification of a global DPI for all images from `topostats/default_config.yaml` to per-plot in
`topostats/plotting_config.yaml` with default of `100` for most values but `800` for existing plots of traces.

Updates validation rules and documentation and I've taken the opportunity to add missing Description to two rows in the
documentation covering the `summary_config.yaml`.
closed 2023-09-13 by ns-rse