AFM-SPM/TopoStats #647: Fork libasd and release to PyPI

ns-rse, 2023-08-04:
THe builds for [libasd](https://github.com/ToruNiina/libasd) are not correct on PyPI and despite contacting the author multiple times we haven't seen this resolved.

We want to support `.asd` format and so have decided that we will fork the repository, rename and get the releases working so that libasd is installable across architectures and we can support `.asd`.

ns-rse, 2023-09-11:
Work on this has been started by @SylviaWhittle and we have [AFM-SPM/topo-libasd](https://github.com/AFM-SPM/topo-libasd). However releasing to PyPI has been a little challenging.

We've migrated configuration via `pyproject.toml` and to dynamic versioning via `setuptools_scm` rather than explicitly in the configuration.

Currently though we hit a problem with uploading to Test PyPI which rejected the built Linux wheels because they were not considered general enough to work on any distribution.

Short term we have therefore opted to make OSX and Linux builds only `sdist` (i.e. source based, as is the case with OSX ARM architecture) as most distributions will have the toolchain required to install.

Longer term though we should implement [manylinux](https://github.com/pypa/manylinux) builds. This builds the Linux wheels on old distros (under Docker containers) so they are as general as possible. We can look at [pyflamegpu](https://github.com/FLAMEGPU/FLAMEGPU2/blob/master/.github/workflows/Manylinux2014.yml) action to inform building.

A current stumbling block for the Windows wheels though is that the package name is not correctly recognised under Python 3.8 and 3.9 and is `UNKNOWN` and so uploading is prevented as this is not a valid package name for PyPI. This may be related to backwards compatibility/lack of support for `pyproject.toml` but I'm unsure at the moment.

SylviaWhittle, 2023-09-29:
I have since given up on building & releasing due to lots of issues building the C++ code for various systems. I have instead written `topofileformats` which decodes the `.asd` files in pure Python.

Long term I plan to incorporate this into `[afmformats](https://github.com/AFM-analysis/afmformats)` as I have been in contact with the dev of the project and he welcomes it, however in the interests of speed, for now it's its own package.

PR #675 (2023-12-19): Add .asd file support
Closes #647 
Closes #304 

This PR adds support for loading `.asd` files into TopoStats, processing all the frames automatically and saving each frame as an individual image output.

This is not going to be the final workflow for `.asd` files, however I want to split this task up into smaller PRs to get feedback from users on how it should be improved going forward.

The actual loading of the `.asd` file is done using my [topofileformats](https://github.com/AFM-SPM/topofileformats) package that I wrote to decode and load `.asd` files into numpy arrays.

## Technical detai [... cut]
closed 2023-12-19 by ns-rse