AFM-SPM/TopoStats #674: No nm scaling when no config file used

derollins, 2023-09-26:
## Describe the bug

When topostats is run with the basic command `topostats process` without a config file the images come out with pixel scale bars (labelled as nm) rather than the physical dimensions of the image.

## Copy of the output

[image]

## To Reproduce

Ran on the test folder with `minicircles.spm`

## TopoStats version

``` bash
Installed version of TopoStats: 2.1.2.dev25+g3a21ec1d```

## Operating System and Python Version

### Operating System

Please let us know what operating system you are using, if you have used more than one then tick all boxes.

* [x] Windows
* [ ] MacOS Intel (pre-2021)
* [x] MacOS M1/M2 (post-2021)
* [ ] GNU/Linux (please include distribution)

### Python Version

Please let us know the version of Python you are using, paste the results of `python --version`

``` bash
Python 3.10.9```

ns-rse, 2023-10-11:
This may have been fixed by @SylviaWhittle in #658 @derollins.

I've run `minicircles.spm` recently (post #658 being merged) and done some checking and the `pixel_to_nm_scaling` value extracted from the `minicircles.spm` is `0.4940029296875`. The image is 1000 x 1000 pixels and so the scaling of the dimensions gives a range of `0` to `515.859` on scaling to `nm` which is I think correct (and so your original image is actually in nanometers rather than pixels).

If you've a moment to check and let us know that would be great please.

Thanks,

@ns-rse

ns-rse, 2023-10-13:
Closing as I'm happy that the `pixel_to_nm_scaling` is being used and the scale is `nm` and not `pixels`.

closed 2023-10-13 by ns-rse