AFM-SPM/TopoStats #679: Violin plots changed as of Seaborn 0.13. [Tests broken].

SylviaWhittle, 2023-10-04:
In making PR #678 , I believe I have discovered a bug in the testing suite & local testing if the Seaborn package is updated.

I was messing about with plotting as I had found a bug caused by an exception thrown by Seaborn regarding KDE plots, and I realised I had not updated Seaborn to check if the issue was resolved.

After updating Seaborn, I found there were a new couple of tests that failed, namely:
- `tests/test_plotting.py::test_plot_violin`
- `tests/test_plotting.py::test_plot_violin_multiple_directories`

The versions of Seaborn in question are: `0.12` and `0.13`.

The differences in the violin plots are:
Multiple:
Old:
[image]
New:
[image]

Single:
Old:
[image]
New:
[image]

I haven't yet checked online to see if others are experiencing this issue or if `0.13` is a breaking release.

We could maybe fix this by just manually adding a legend in the code and updating the test plots. I will try this tomorrow, as I have been looking into this too long already.

Alternatively we could fix the version for Seaborn at `0.12` and descend further into dependancy hell.

The test suite in GitHub Actions fails too, as you can see in the presenting symptom: https://github.com/AFM-SPM/TopoStats/actions/runs/6399786751/job/17372413989?pr=678. Which would need fixing in version too if we went that route.

Perhaps I'm just being silly and it's a trivial matter though!

Thoughts, @ns-rse ?

Branch: Main, Current, 2.1.2.dev41+gda58094ab
OS: MacOS, Ubuntu, Windows
Config: Default
Python Versions: 3.8, 3.9, 3.10, 3.11 (Courtesy of GH-A) 

Here is a clean log of evidence for the issue on unaltered `main`:
```bash
    ~/Documents/TopoStats    main *23 ?1                                                                                                                                                                                                        new   01:50:37  
❯ pip install seaborn==0.12  
Collecting seaborn==0.12
  Downloading seaborn-0.12.0-py3-none-any.whl (285 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 285.1/285.1 kB 2.4 MB/s eta 0:00:00
Requirement already satisfied: numpy>=1.17 in /opt/homebrew/Caskroom/miniforge/base/envs/new/lib/python3.10/site-packages (from seaborn==0.12) (1.23.4)
Requirement already satisfied: pandas>=0.25 in /opt/homebrew/Caskroom/miniforge/base/envs/new/lib/python3.10/site-packages (from seaborn==0.12) (1.5.3)
Requirement already satisfied: matplotlib>=3.1 in /opt/homebrew/Caskroom/miniforge/base/envs/new/lib/python3.10/site-packages (from seaborn==0.12) (3.7.1)
Requirement already satisfied: contourpy>=1.0.1 in /opt/homebrew/Caskroom/miniforge/base/envs/new/lib/python3.10/site-packages (from matplotlib>=3.1->seaborn==0.12) (1.0.7)
Requirement already satisfied: cycler>=0.10 in /opt/homebrew/Caskroom/miniforge/base/envs/new/lib/python3.10/site-packages (from matplotlib>=3.1->seaborn==0.12) (0.11.0)
Requirement already satisfied: fonttools>=4.22.0 in /opt/homebrew/Caskroom/miniforge/base/envs/new/lib/python3.10/site-packages (from matplotlib>=3.1->seaborn==0.12) (4.39.2)
Requirement already satisfied: kiwisolver>=1.0.1 in /opt/homebrew/Caskroom/miniforge/base/envs/new/lib/python3.10/site-packages (from matplotlib>=3.1->seaborn==0.12) (1.4.4)
Requirement already satisfied: packaging>=20.0 in /opt/homebrew/Caskroom/miniforge/base/envs/new/lib/python3.10/site-packages (from matplotlib>=3.1->seaborn==0.12) (23.0)
Requirement already satisfied: pillow>=6.2.0 in /opt/homebrew/Caskroom/miniforge/base/envs/new/lib/python3.10/site-packages (from matplotlib>=3.1->seaborn==0.12) (9.4.0)
Requirement already satisfied: pyparsing>=2.3.1 in /opt/homebrew/Caskroom/miniforge/base/envs/new/lib/python3.10/site-packages (from matplotlib>=3.1->seaborn==0.12) (3.0.9)
Requirement already satisfied: python-dateutil>=2.7 in /opt/homebrew/Caskroom/miniforge/base/envs/new/lib/python3.10/site-packages (from matplotlib>=3.1->seaborn==0.12) (2.8.2)
Requirement already satisfied: pytz>=2020.1 in /opt/homebrew/Caskroom/miniforge/base/envs/new/lib/python3.10/site-packages (from pandas>=0.25->seaborn==0.12) (2022.7.1)
Requirement already satisfied: six>=1.5 in /opt/homebrew/Caskroom/miniforge/base/envs/new/lib/python3.10/site-packages (from python-dateutil>=2.7->matplotlib>=3.1->seaborn==0.12) (1.16.0)
Installing collected packages: seaborn
  Attempting uninstall: seaborn
    Found existing installation: seaborn 0.13.0
    Uninstalling seaborn-0.13.0:
      Successfully uninstalled seaborn-0.13.0
Successfully installed seaborn-0.12.0
    ~/Documents/TopoStats    main *23 ?1                                                                                                                                                                                                        new   01:50:49  
❯ pytest tests/test_plotting.py
============================================================================================================================ test session starts =============================================================================================================================
platform darwin -- Python 3.10.9, pytest-7.2.2, pluggy-1.0.0
[... 213 lines cut ...]
topostats/logs/logs.py                 25      0   100%
topostats/plotting.py                 144     13    91%
topostats/plottingfuncs.py            115     95    17%
topostats/processing.py               213    189    11%
topostats/run_topostats.py            113     94    17%
topostats/scars.py                    105     94    10%
topostats/statistics.py                26     20    23%
topostats/theme.py                     40     23    42%
topostats/thresholds.py                31     18    42%
topostats/tracing/dnacurvature.py      24     24     0%
topostats/tracing/dnatracing.py       413    364    12%
topostats/tracing/skeletonize.py       23     12    48%
topostats/tracing/tracingfuncs.py     613    550    10%
topostats/utils.py                     76     44    42%
topostats/validation.py                15      5    67%
-------------------------------------------------------
TOTAL                                3032   2378    22%

========================================================================================================================== short test summary info ===========================================================================================================================
FAILED tests/test_plotting.py::test_plot_violin
FAILED tests/test_plotting.py::test_plot_violin_multiple_directories
======================================================================================================================== 2 failed, 18 passed in 5.24s ========================================================================================================================
    ~/Documents/TopoStats    main *23 ?1                                                                                                                                                                                                 7s   new   01:51:24  
❯ 
```

ns-rse, 2023-10-04:
Noticed this on the tests run on `pre-commit-autoupdate-ci`too.

Looks like a change in the defaults for Seaborn and a correction/improvement to multiple plotting (in our test image the violins don't align with the marks on the axes).

I'm happy to take this (need something simple to ease back into coding) and think updating our test images and setting a default to include a legend in multiple violin plots would be sensible.

Once merged the pre-commit and your #678 can be updated to `main` and they should then pass the tests and we can make a bug fix release.

SylviaWhittle, 2023-10-04:
Thanks @ns-rse !

commit 21b20df4 (2023-10-04): Legend plotted on single image violin plots; update to Seaborn 0.13

Closes #679

Newer version of Seaborn corrected alignment of violin plots when the x-axis had multiple categories which broke
tests (thanks for the detailed investigation @SylviaWhittle :+1: )

Taken the opportunity to ensure the legend is only plotted when multiple images/directories are plotted.
closed 2023-10-04 by ns-rse