AFM-SPM/TopoStats #694: Move Notebooks to run on JupyterLite

ns-rse, 2023-10-11:
I've started the [AFM-SPM/jupyterlite-topostats](https://github.com/AFM-SPM/jupyterlite-topostats/) repository with a view to provisioning Notebooks via [JupyterLite](https://github.com/jupyterlite/jupyterlite) which uses [WebAssembly](https://webassembly.org) to run a Jupyter server completely in a users browser, no need to setup and install a virtual environment.

Most work will be tracked via issues/features and undertaken in the [AFM-SPM/jupyterlite-topostats](https://github.com/AFM-SPM/jupyterlite-topostats/) repository but it may be prudent to write and include some features for reading and making files available under `topostats.io` module. They will be documented here when the time arises and once everything is working well this issue can be closed by the removal of Notebooks.

Currently the notebooks are in place and load, but there is a dependency issue because there are no Python wheels available for Numpy-1.26.0 (see [issue](https://github.com/pyodide/pyodide/issues/4213)) which means that TopoStats itself will not install in the Notebooks.

+ [ ] Update documentation to point to using [jupyterlite-topostats](https://afm-spm.github.io/jupyterlite-topostats/lab/index.html)
+ [ ] Remove Notebooks from TopoStats.

ns-rse, 2024-11-19:
Not sure if this will be require din light of #987 would people be happy to close this in favour of getting the Notebooks running Colab?

/cc @SylviaWhittle @MaxGamill-Sheffield @llwiggins @alicepyne

alicepyne, 2024-11-19:
Yep!

closed 2024-11-20 by ns-rse