AFM-SPM/TopoStats #696: Dnatracing fails when one molecule spline fails *important*

SylviaWhittle, 2023-10-12:
## Checklist

The issue:
It appears that there may be a little bit more work to be done on the removing loops from `dnatracing.py`. There is an error where in `get_splined_traces()`, where if a `ValueError` is raised on line `370`, then `self.mol_is_circular.pop(dna_num)` is called. This is code from when the `dnaTrace` class acted on the whole image (all molecules) and not a single molecule. Hence, `dna_num` is undefined, `self.mol_is_circular` is a `bool` not a `list` and `.pop()` is therefore not defined for `self.mol_is_circular`.

There are lines below it that will be affected similarly. I tried to have a go at fixing this but it seems like it will need a bit of an in-depth consideration as I'm not familiar with the control flow surrounding this, and how to represent a failure state properly without breaking the data structures.

Then again, I believe this fix might not be necessary if I tidy up and address the comments on #653 Improve Linear Splining, as that removes this block of code entirely. Thoughts would be appreciated @ns-rse 😄 

Version: Current main branch: 2.1.3.dev2+g6354f5477.d20231012
Config file: Default
Processing file: minicircle.spm
Command: `pytest tests/test_processing.py::test_run_dnatracing` (With `run_dnatracing()` in `processing.py` set to **raise** errors, not ignore them)
Python Version: 3.10
OS: MacOS

Output:
```bash
self = <topostats.tracing.dnatracing.dnaTrace object at 0x163b3cd00>

    def get_splined_traces(self):
        """Gets a splined version of the fitted trace - useful for finding the radius of gyration etc
    
        This function actually calculates the average of several splines which is important for getting a good fit on
        the lower res data"""
    
        step_size = int(7e-9 / (self.pixel_to_nm_scaling))  # 3 nm step size
        interp_step = int(1e-10 / self.pixel_to_nm_scaling)
        # Lets see if we just got with the pixel_to_nm_scaling
        # step_size = self.pixel_to_nm_scaling
        # interp_step = self.pixel_to_nm_scaling
    
        self.splining_success = True
        nbr = len(self.fitted_trace[:, 0])
    
        # Hard to believe but some traces have less than 4 coordinates in total
        if len(self.fitted_trace[:, 1]) < 4:
            self.splined_trace = self.fitted_trace
            # continue
    
        # The degree of spline fit used is 3 so there cannot be less than 3 points in the splined trace
        while nbr / step_size < 4:
            if step_size <= 1:
                step_size = 1
                break
            step_size = -1
        if self.mol_is_circular:
            # if nbr/step_size > 4: #the degree of spline fit is 3 so there cannot be less than 3 points in splined trace
    
            # ev_array = np.linspace(0, 1, nbr * step_size)
            ev_array = np.linspace(0, 1, int(nbr * step_size))
    
            for i in range(step_size):
                x_sampled = np.array(
                    [self.fitted_trace[:, 0][j] for j in range(i, len(self.fitted_trace[:, 0]), step_size)]
                )
                y_sampled = np.array(
                    [self.fitted_trace[:, 1][j] for j in range(i, len(self.fitted_trace[:, 1]), step_size)]
                )
    
                try:
                    tck, u = interp.splprep([x_sampled, y_sampled], s=0, per=2, quiet=1, k=3)
                    out = interp.splev(ev_array, tck)
                    splined_trace = np.column_stack((out[0], out[1]))
                except ValueError:
                    # Value error occurs when the "trace fitting" really messes up the traces
    
                    x = np.array(
                        [self.ordered_trace[:, 0][j] for j in range(i, len(self.ordered_trace[:, 0]), step_size)]
                    )
[... 147 lines cut ...]
topostats/filters.py                  154    133    14%
topostats/grains.py                   105     81    23%
topostats/grainstats.py               331    281    15%
topostats/io.py                       386    320    17%
topostats/logs/logs.py                 25      0   100%
topostats/plotting.py                 156    127    19%
topostats/plottingfuncs.py            115     41    64%
topostats/processing.py               229    169    26%
topostats/run_topostats.py            113    113     0%
topostats/scars.py                    105     94    10%
topostats/statistics.py                26     20    23%
topostats/theme.py                     40     11    72%
topostats/thresholds.py                31     18    42%
topostats/tracing/dnacurvature.py      24     24     0%
topostats/tracing/dnatracing.py       435    162    63%
topostats/tracing/skeletonize.py       23     12    48%
topostats/tracing/tracingfuncs.py     613    231    62%
topostats/utils.py                     76     58    24%
topostats/validation.py                15     15     0%
-------------------------------------------------------
TOTAL                                3083   1991    35%

===================================================================================================================================== short test summary info =====================================================================================================================================
FAILED tests/test_processing.py::test_run_dnatracing - AttributeError: 'bool' object has no attribute 'pop'
```

ns-rse, 2023-10-13:
On first look at this I think all of the `*.pop()` can be removed.

They were `pop()` because each of the classes properties was a list (perhaps dictionary, can't remember but suspect list) as the class was iterating through each of the grains, but if a `ValueError` was encountered then it failed to trace and so needed removing.

I think, but would have to dig deeper and check, that these attributes (`self.mol_is_circular` / `self.disordered_trace` / `self.grain` / `self.ordered_trace`) could remain as is and we just set `self.splining_success = False` here. (I suspect that is too simple a solution though and its more complicated!).

ns-rse, 2023-10-13:
Also, if removing these lines works, it would be worth writing (yet another) test to cover this `try: ... except:`.

SylviaWhittle, 2023-10-13:
Thanks @ns-rse 

Do you have thoughts about perhaps merging (one ready) #653 to enable us to delete this block of code?

Also, removing the `.pop` doesn't appear to work as a new error is thrown: the variable `spline_running_total` undefined.

I have just realised though that I don't think `spline_running_total` is ever defined...

ns-rse, 2024-04-03:
Possible duplicate of #366

MaxGamill-Sheffield, 2024-04-03:
@ns-rse These lines aren't in the code anymore so is this ok to be closed? Doesn't seem like a duplicate as this is one specific cause of #366 whereas #366 is surrounding the try/catch.

ns-rse, 2024-04-03:
@MaxGamill-Sheffield Agreed on closer inspection this is a possible cause of #366.

The broad `try:...except:` is still in place on the `main` branch in [`topostats.processing.py`](https://github.com/AFM-SPM/TopoStats/blob/6d24aaa2f100a67748b4349d1b226a868db6f618/topostats/processing.py#L400) and when it fails [grain statistics are returned](https://github.com/AFM-SPM/TopoStats/blob/6d24aaa2f100a67748b4349d1b226a868db6f618/topostats/processing.py#L480) which means users will at least get _something_ even if tracing fails (note however that you have disabled this `try: ... except:` on branch `maxgamill-sheffield/800-better-processing` see [line 400](https://github.com/AFM-SPM/TopoStats/blob/1918e908cb73d3d4836083b74135420e8d5be8d5/topostats/processing.py#L400) and [line 480](https://github.com/AFM-SPM/TopoStats/blob/1918e908cb73d3d4836083b74135420e8d5be8d5/topostats/processing.py#L488) which I only know about because PyLint complained about the use of `"""..."""` to comment a section as it made it look like the `return ...` wasn't at the end of the function, better to use `# ` for commenting regions).

I've not looked at this in a long time so can't say right now what to do, however, with the desire to get all of the revised tracing/pruning/detangling into the `main` branch so that papers can be published and new versions released I would be inclined to stay focused on that aspect for now and come back to this once that work is completed. 

Removing the broad `try: ...except:`, which was thrown in because of a desire to have something working quickly if I remember correctly, could perhaps be the next Milestone for TopoStats and would encapsulate this issue.

I personally don't have much of a problem with issues being noted, they are at least then documented and can be dealt with in order of priority but I am wary of assuming things are fixed without checking and don't have masses of capacity to investigate this myself right now given the above priority and the height profile work.

SylviaWhittle, 2024-11-13:
DNAtracing is completely overhauled and so closing as this is taken care of

closed 2024-11-13 by SylviaWhittle