AFM-SPM/TopoStats #705: TopoStats Logo

ns-rse, 2023-10-20:
With talk of branding TopoStats plots I thought it would be useful if we developed a logo for the software.

This helps give a brand identity to the software, it can be added to the documentation web-site, used as a logo here on GitHub and we can make images readily available for inclusion in slides that are presented as hyperlinks so that when people share and view the slides (e.g. if they are written in Quarto and hosted on GitHub pages) they can easily click through and find the repository and other information.

As the image may be rendered in quite small sizes it needs to have a bold outline. Perhaps one of the grains/loops from `minicircles.spm` might be appropriate. For example the following are scaled to 100, 50 and 25 pixels in width

[image]

[image]

[image]

Original...

[image]

llwiggins, 2023-10-20:
I really like this idea! I wondered if it would also be nice to add a trace overlay to the molecule to highlight the quantitative nature of TopoStats?

ns-rse, 2023-10-20:
Could do, my concern would be visibility when made small, the 25 pixel image you can just about see the molecule but I don't think a trace would be visible.

I'm not at all adverse to a more comic style logo or some sort of Hex image as is commonly used for other packages, but I've close to zero if not negative design skills so went with the easy option.

MaxGamill-Sheffield, 2023-10-27:
Maybe we can find a fat-tip image to use as the DNA should be thicker on the image? Then we might be able to have a trace overlay too

ns-rse, 2023-11-10:
Any more opinions on this?

I think given the need to render the image in a very small size any tracing lines will be barely visible. As an example look at the AFM-SPM organisation logo at the top of this page, its not really possible to discern what things are.

MaxGamill-Sheffield, 2023-11-10:
What about cropping and combining 2 images of a T and a S looking molecules? There's some T ones in the high supercoiling minicircle datasets and I assume some S looking linear strands somewhere. When I start getting into the minicircle stuff I can take a look