AFM-SPM/TopoStats #709: Move cropped images to their own folder

SylviaWhittle, 2023-10-26:
It has been requested that we have a "cropped_images" folder alongside the "processed" folder for easy access to molecule crops when `image_set` is set to all.

All the other file-specific images such as the cropped masks would stay where they are, this would just be for the actual cropped images, to allow for easy comparison of molecules in the whole dataset.

It was seemed insufficient to use a command such as:
```bash
mkdir ./cropped_grains; cp ./**/*_grain_image_*.png ./cropped_grains
```
due to the need for the default output to be simple and understandable for new users without having to use commands.

This should be a simple change, and perhaps could be a config setting, for example:
```yaml
save_cropped_images_separately = True # Whether to save cropped images in their own folder or with all the other grain specific images.
```

ns-rse, 2023-10-27:
This looks like a duplicate of #702, could we close one or the other please?

I like the idea of making this configurable though. :+1:

SylviaWhittle, 2023-10-27:
Oops, sorry. I made this during a meeting and didn't realise Max already made one :)

closed 2023-10-27 by SylviaWhittle