AFM-SPM/TopoStats #714: Remove return of scars_mask from remove_scars()

ns-rse, 2023-11-03:
In reviewing #711 it was identified that there is no use for the returned `first_marked_mask` as it is not used in any subsequent processing. Commit [#1ffe0c](https://github.com/AFM-SPM/TopoStats/pull/711/commits/1ffe0cd17403e3d00fa1eaefa73a5c71aa48edda) therefore assigns it to `_` which is discarded.

Rather than return something and discard it we should perhaps consider removing the mask from being returned at all and changing [line 434](https://github.com/AFM-SPM/TopoStats/blob/44877cbdf9117e68122e6bc7cdcee5b6b528c4cb/topostats/scars.py#L434) to...

```python
    return img
```

As an aside I noticed the typehint for the `remove_scars()` function doesn't have an explicit return type which if this change is made would be `npt.NDArray` (I discovered Numpy now has a specific [numpy.typing](https://numpy.org/devdocs/reference/typing.html) module which if `import numpy.typing as npt` would allow use of the generic [`NDArray`](https://numpy.org/devdocs/reference/typing.html#numpy.typing.NDArray)). The docstring also says `self.img` is returned when it should, after this change, just be...

```
Returns
-------
npt.NDArray
    ...
```

SylviaWhittle, 2023-11-03:
So I think we do use the returned scar mask in the second instance, where one of the images for the `Filters`' `images` dictionary is set to `"scar_mask"`:
[image]

But it isn't plotted even with 'all' as the image set.

When developing this feature I had thought it would be of interest to users to see the scars that the scar removal had removed, however given that it was never plotted and that nobody has asked / complained about it, perhaps it should be removed?

I personally have used this a couple times, but not more than that, and as a developer, it's simple to add it back in when needed.

Less code is better

ns-rse, 2023-11-07:
Less code is better, the balance for me is how much demand there is for the scar masks. :balance_scale:

ns-rse, 2023-11-10:
What do you think about keeping these arrays or not? I noticed mention in #675 about processing `.asd` taking up ~2GB of RAM so keeping these would only inflate that.