AFM-SPM/TopoStats #748: Add horizontal grain profiles

SylviaWhittle, 2023-12-05:
**Is your feature request related to a problem? Please describe.**
It has been requested that topostats have a feature similar to Gwyddion where the horizontal (in the image) height profile of a grain would be taken. In other words, a horizontal line is placed over the molecule, and the pixels that form the line have their heights measured to form a profile for that molecule.

**Describe the solution you'd like**
During grainstats, a horizontal, one pixel thick line of pixels should be formed, with the length sufficient to cover the entire molecule from left to right, and the line intersecting the centroid of the molecule. The heights for those pixels should be obtained and saved as a line plot. Probably only saved with image_set all.

While it does seem arbitrary to just take one line through the centre, and we will miss a lot of information, this is is not an issue, as this line profile is just to get an idea of the height profile of the molecule.

slackline, 2023-12-05:
Ultimately the profile of any horizontal line is available as the Numpy array for the given grain so its "just" a case of finding the centroid of a grain (and skimage has functions for doing this) and extracting that row and the plotting.

Some random questions off the top of my head (not necessarily directed at you @SylviaWhittle ).

At what point in the processing should we extract this information, is it from the untraced or traced grain? I suspect the former but would like to check

What should be done for linear molecules?

In the future is any other profile likely to be required other than that which passes through the centroid? If so will these always be horizontal or are other angles/paths required such as the profile through the widest part of a circular molecule[^1]?

[^1]: I would be surprised if the orientation of a molecule is always as desired in the horizontal plane and is essentially random.

alicepyne, 2023-12-05:
> Currently the classification of grains into circle or linear is done during tracing and there is no classification into blob.
>
> A simple solution would be to just plot the required profile regardless of type, even if its meaningless, and that can be done, I was just curious at which stage the cross-section should be taken during the processing pipeline, for example after applying a Gaussian blur the values may be changed.
>
> What is the intended use of the data? I ask as I'm curious about why only the horizontal axis is required.
>
> These are the same grain with the same heights, but orientated differently.
>
> 0 0 0 0 0
> 0 1 1 1 0
> 0 1 2 1 0
> 0 1 2 1 0
> 0 1 2 1 0
> 0 1 2 1 0
> 0 1 2 1 0
> 0 1 2 1 0
> 0 1 1 1 0
> 0 0 0 0 0
> 0 0 0 0 0 0 0 0 0 0
> 0 1 1 1 1 1 1 1 1 0
> 0 1 2 2 2 2 2 2 1 0
> 0 1 1 1 1 1 1 1 1 0
> 0 0 0 0 0 0 0 0 0 0
> ...and you'd get very different profiles from the horizontal only.
>
> What information is sought/trying to be extracted?
> Is there some sort of step prior to or during scanning that ensures grains are orientated in a specific way?
> minicircle.spm looks fairly random to me but I'm very wary of generalising based on a single sample.

I think this would be an option - like “crop” in the config file, Where if the suer wants to they would plot a line of “user defined length” or through the centre of the blob from the centre only. This is to use as single examples alongside the numbers obtained from topostats e.g. height, are volume, partly for sense checking, partly for visualisation. 

If you did this over all the molecules, you would get an average. Which would show you heterogeneity you display below.

ns-rse, 2023-12-05:
Wouldn't you have heterogeneous measurements across a single molecule depending on the orientation though?

Would it perhaps be useful to use a slice along the line parallel to the minimum feret diameter which passes through the centroid as that would then pick up the height profile along (roughly[^1]) the longest slice?

I'm probably over thinking this though so please do say but I'm trying to anticipate other scenarios and think about general solutions at the outset.

[^1]: I say roughly because there is no guarantee the extremes will always be through the centroid

alicepyne, 2023-12-05:
> Footnotes

You could do that, open to measuring the line along the max feret length
through the centroid, or through the max and min feret as a sort of cross.

ns-rse, 2023-12-05:
Further notes after todays meeting with @SylviaWhittle , @llwiggins and @MaxGamill-Sheffield 

+ Two profiles passing through the centroid and aligning with the orientation of the minimum and maximum feret sound like the way to go.
+ Returning the raw data as well as making a plot would be sensible as it is then available for further analysis by end users.

This feature request should take priority over writing tests for the [maxgamill-sheffield/cats](https://github.com/AFM-SPM/TopoStats/tree/maxgamill-sheffield/cats) branch which underpins the forthcoming paper which is due to be finalised 2024-01.

ns-rse, 2023-12-07:
Writing my thoughts down here before coding...

## Algorithm

1. Calculate feret distances
2. Extract co-ordinates of minimum and maximum feret distances
3. Rotate image so that horizontal aligns with minimum feret distance/coordinates and extract profile
4. Rotate image so that horizontal aligns with maximum feret distance/coordinates and extract profile

## How to do this

Scikit-image has methods/code for calculating feret distance (see [`skimage.measure._regionprops.py`](https://github.com/scikit-image/scikit-image/blob/604d0776f882e2ba6c50a2b1fc254d718edb659a/skimage/measure/_regionprops.py#L495) which we can extract out to a function...

```python
from typing import Tuple
import numpy as np
import numpy.typing as npt

def calculate_feret(image: npt.NDArray) -> Tuple[npt.NDArray, npt.NDArray]:
    # Import necessary functions from skimage
    from skimage.morphology.convex_hull import convex_hull_image
    from skimage.measure import find_contours, marching_cubes

    image_convex = convex_hull_image(image)
    spacing =
    identity_convex_hull = np.pad(image_convex, 2, mode="constant", constant_values=0)
    if self._ndim == 2:
        coordinates = np.vstack(find_contours(identity_convex_hull, 0.5, fully_connected="high"))
    elif self._ndim == 3:
        coordinates, _, _, _ = marching_cubes(identity_convex_hull, level=0.5)
    distances = pdist(coordinates * self._spacing, "sqeuclidean")
    return distances, coordinates
```

We could then easily extract the co-ordinates for the `min(distances)` and `max(distances)` by 
calculating...

```python
min_feret_coordinates = coordinates[np.argwhere(np.min(distances) == distances],]
max_feret_coordinates = coordinates[np.argwhere(np.min(distances) == distances],]
```

These can be applied to the `regionprops` of a grain as `extra_parameters` in case we want any other statistics on the region or individually but would serve as useful functions to address #750 (which involves a bit more work).

## Questions (mostly to myself)

+ Is rotation actually necessary?
+ Should the profile extend to the very edges of the cropped grain?

ns-rse, 2023-12-07:
Not so straight-forward as the following shows, the `scikit-image` code calculates pairwise distances (courtesy of `scipy.spatial.distance.pdist()` for all points and there will always be some zeros.  Fortunately I found in [this issue](https://github.com/scikit-image/scikit-image/issues/4817) reference to calculating the minimum feret distance for 2-D objects written up in a [gist](https://gist.github.com/VolkerH/0d07d05d5cb189b56362e8ee41882abf) by @VolkerH so will give that a whirl.

ns-rse, 2024-02-13:
See comments in #755 (in particular [here](https://github.com/AFM-SPM/TopoStats/issues/798)).

The algorithm proposed in the [Mathworks article](https://blogs.mathworks.com/steve/2018/02/20/minimum-feret-diameter/) works for everything _except_ triangles and it took me a while (most of yesterday) to work through why, carefully comparing the method @SylviaWhittle implemented in `GrainStats.get_min_max_feret()` and the vectorised approach I'm working on in `measure.feret` sub-module.

| Step        | `get_min_max_feret()` | `get_min_max_feret()` | `feret.min_feret_triangles()` | `feret.min_feret_triangles()` | Match?         |
|:-----------:|:--------------:|:---------------:|:-----------------:|:---------------:|:-------------:|
|             | **Upper Hull** | **Lower Hull**  | **Upper Hull**        | **Lower Hull**      |               |
|             | [[1, 1],       | [[1, 1]         | [[1, 1],          | [[1, 1],        | Yes           |
|             |  [1, 2],       |  [2, 1]]        |  [1, 2],          |  [2, 1]]        |               |
|             |  [2, 1]]       |                 |  [2, 1]]          |                 |               |
| Rotating    |         [[2, 1], [1, 1]]         |            [[2, 1], [1, 1]]         | Not in order but the same set of pairs.  |
| Caliper     |         [[2, 1], [1, 2]]         |            [[2, 1], [1, 2]]         |   |
| Pairs       |         [[1, 1], [1, 2]]         |            [[1, 1], [1, 2]]         |   |
| Itter 1     |                |                 |                   |                 | Same points, different hulls   |
| Base 1st    | [1, 1]         |                 |                   | [1, 1]          |   |
|      2nd    | [1, 2]         |                 |                   | [1, 2]          |   |
| Itter 2     |                |                 |                   |                 | Same points, different hulls   |
| Base 1st    |                | [2, 1]          | [2, 1]            |                 |   |
|      2nd    |                | [1, 1]          | [1, 1]            |                 |   |
| Itter 3     |                |                 |                   |                 | NO!!! This is the problem.        |
| Base 1st    | [1, 2]         |                 | [1, 1]            |                 |    |
|      2nd    | [2, 1]         |                 | [2, 1]            |                 |    |

The clause is because of the condition...

```python
next_index = 1 if index == n_pairs - 1 else index + 1
```
...which on the final iteration means the upper hull is using the last element ([1, 1]) and the first element ([2, 1]) which are the same points on the lower hull which have already been seen as the base. It also means that the base formed by [1, 2] and [2, 1] has not been observed which is why I get the minimum height returned as `1.0` rather than `0.7071067811`.

I can't work out a way of adapting this algorithm so have gone with @SylviaWhittle solution of calculating the minimum feret based on the triangle formed by a rotating caliper pair (of which one point is the apex) and the next adjacent point on one of the hulls (albeit with some tweaks to the code layout).

## Minimum Feret Co-Ordinates

After some revision the stated desire is to have...

> Two profiles passing through the centroid and aligning with the orientation of the minimum and maximum feret sound like the way to go.

This is probably most easily obtained by determining the mid-point between the maximum feret co-ordinates and taking a profile that is prependicular to the maximum feret at this point.

But I can envisage two issues based on the limited sample of images I've seen.

1. The minimum feret might be some distance from the centroid, if there is a "twist" in a loop forming what looks like a figure of eight the profile from taking the profile perpendicular to the mid-point of the maximum feret, whilst consistent across molecules may not be representative/informative of the minimum feret.
2. The minimum feret itself isn't based on the caliper points, but a pair of calipers and an adjacent convex hull point. As such the caliper doesn't represent the co-ordinates of the minimum feret, the apex and the mid-point between adjacent hull points does. This can be calculated exactly but will on most occassions not align to co-ordinates in the numpy array and so an approximation would have to be used to get those co-ordinates.

For 1) I think perhaps returning both the suggested smaller profiles might be helpful, and a clear description of what these represents should be included in the documentations data dictionary that describes the grain statistics that are returned.

For 2) I propose taking the nearest co-ordinate to the mid-point of the base of the triangle formed in calculating the minimum feret distance and returning the profile between this and the apex of the triangle.

A very crude schematic to maybe aid with visualisation (would take too long for me to plot this out).

```

--- = parallel lines forming minimum feret
~~~ = skeleton (jelly bean like shape)
|   = Minimum feret distance
*   = Antipodal Calipers
@   = Third point on convex hull forming triangle 
+   = Point to be estimated

-------------------------------
             ~*~
         ~~~~ | ~~~~
      ~~~    /|\     ~~~
   ~~       / | \       ~~ 
  ~        /  |  \        ~
  ~~      / ~~|~~ \      ~~ 
    ~~~~~@~~  +  ~~*~~~~~
-------------------------------
```

PR #871 (2024-07-17): Incorporating height profile into GrainStats
Closes #748 

After the protracted wrestling I had of how to get minimum feret distances out (see #755) I've decided to finish this work off now as I'm unlikely to look at it again until September and had been tinkering with it over the last week and it is more efficient to finish it off now than struggle to remember where I was in two months time.

This PR includes two commits, importantly one fixes a problem with the `topostats.measure.height_profiles.interpolate_height_profiles()` which was introduced in #859 (details on that below). 

## Height  [... cut]
closed 2024-07-17 by ns-rse