AFM-SPM/TopoStats #785: [Bug]: TopoStats error on yaml dump due to depreciation

MaxGamill-Sheffield, 2024-01-23:
### Checklist

- [X] Re-run analysis with `topostats process --core 1`.
- [X] Describe the bug.
- [X] Include the configuration file.
- [X] Copy of the output.
- [X] The exact command that failed. This is what you typed at the command line, including any options.
- [X] TopoStats version, this is reported by `topostats --version`
- [X] Operating System and Python Version

### Describe the bug

When using `pip install topostats` in a new conda environment, all is installed ok. But when I run the test file using `topostats process`, it fails at the very end with the below error:

### Copy of the output

[image]

### Include the configuration file

This is the `default_config.yaml` file as one is not being specified.

### To Reproduce

Create a new environment with python (I used 3.11). Seems to be an issue with the imported library `run-yaml` as version 0.18.* fails but 0.17.* works.

### TopoStats Version

2.1.2

### Python Version

3.11

### Operating System

MacOS M1/M2 (post-2021)

### Python Packages

asttokens==2.4.1
contextlib2==21.6.0
contourpy==1.2.0
cycler==0.12.1
decorator==5.1.1
executing==2.0.1
fonttools==4.47.2
h5py==3.10.0
igor2==0.5.3
imageio==2.33.1
ipython==8.20.0
jedi==0.19.1
joblib==1.3.2
kiwisolver==1.4.5
lazy_loader==0.3
matplotlib==3.8.2
matplotlib-inline==0.1.6
networkx==3.2.1
numpy==1.26.3
packaging==23.2
pandas==2.2.0
parso==0.8.3
pexpect==4.9.0
pillow==10.2.0
prompt-toolkit==3.0.43
psutil==5.9.8
ptyprocess==0.7.0
pure-eval==0.2.2
pyfiglet==1.0.2
Pygments==2.17.2
pyparsing==3.1.1
pyspm==0.6.0
python-dateutil==2.8.2
pytz==2023.3.post1
PyYAML==6.0.1
ruamel.yaml==0.18.5
ruamel.yaml.clib==0.2.8
schema==0.7.5
scikit-image==0.22.0
scikit-learn==1.4.0
scipy==1.12.0
seaborn==0.13.1
six==1.16.0
stack-data==0.6.3
threadpoolctl==3.2.0
tifffile==2023.12.9
topostats==2.1.2
tqdm==4.66.1
traitlets==5.14.1
tzdata==2023.4
wcwidth==0.2.13

MaxGamill-Sheffield, 2024-01-23:
This is an issue only with the pip installable version, not with the current version @ns-rse can we make a release after the cmap PR is sorted? Seems like you already fixed this previously @ns-rse :)

ns-rse, 2024-01-23:
Probably #450 by the sounds of it?

Bug fix release can definitely be made, happy to include #777 if you don't mind giving that PR a the rubber stamp I'll do so tomorrow (just in time for the Computer Science Newsletter where I include new software releases :smile: )

closed 2024-01-23 by ns-rse