AFM-SPM/TopoStats #795: [Bug]: phantom traces

LHolmes2, 2024-02-07:
### Checklist

- [X] Re-run analysis with `topostats process --core 1`.
- [X] Describe the bug.
- [X] Include the configuration file.
- [X] Copy of the output.
- [X] The exact command that failed. This is what you typed at the command line, including any options.
- [X] TopoStats version, this is reported by `topostats --version`
- [X] Operating System and Python Version

### Describe the bug

The traces appear in the wrong X,Y coordiantes. 

### Copy of the output

[image]

### Include the configuration file

# Configuration from TopoStats run completed : 2024-02-05 12:10:34
# For more information on configuration and how to use it:
# https://afm-spm.github.io/TopoStats/main/configuration.html
base_dir: /Users/libbyholmes/Library/CloudStorage/GoogleDrive-epholmes1@sheffield.ac.uk/Shared
  drives/Pyne Group/AFM_Data/Telomeres/20240201_pICOz_TEL12_shelterin
cores: 2
dnatracing: {min_skeleton_size: 10, pad_width: 1, run: true, skeletonisation_method: topostats,
  spline_circular_smoothing: 0.0, spline_linear_smoothing: 5.0, spline_step_size: 7e-09}
file_ext: .spm
filter:
  gaussian_mode: nearest
  gaussian_size: 0.0
  otsu_threshold_multiplier: 1.0
  remove_scars: {max_scar_width: 4, min_scar_length: 16, removal_iterations: 2, run: false,
    threshold_high: 0.666, threshold_low: 0.25}
  row_alignment_quantile: 0.5
  run: true
  threshold_absolute: {above: 1.0, below: -1.0}
  threshold_method: std_dev
  threshold_std_dev: {above: 1.0, below: 10.0}
grains:
  absolute_area_threshold:
    above: [300, 3000]
    below: [null, null]
  direction: above
  otsu_threshold_multiplier: 1.0
  remove_edge_intersecting_grains: true
  run: true
  smallest_grain_size_nm2: 50
  threshold_absolute: {above: 1.0, below: -1.0}
  threshold_method: std_dev
  threshold_std_dev: {above: 1.0, below: 10.0}
grainstats: {cropped_size: 40.0, edge_detection_method: binary_erosion, run: true}
loading: {channel: Height}
log_level: info
output_dir: /Users/libbyholmes/Library/CloudStorage/GoogleDrive-epholmes1@sheffield.ac.uk/Shared
  drives/Pyne Group/AFM_Data/Telomeres/20240201_pICOz_TEL12_shelterin
plotting:
  axes: true
  colorbar: true
  histogram_log_axis: false
  image_set: core
  mask_cmap: blu
  num_ticks: [null, null]
  pixel_interpolation: null
  run: true
  save_format: png
  style: topostats.mplstyle
  zrange: [-2, 3]
summary_stats: {config: null, run: true}

### To Reproduce

Run on main. File found in Shared drives/Pyne Group/Code/Topostats Error Files/Phantom traces/20240201_8ng_topo_picoz_tel12__KCLshelterinbuffer_mgni_eph.0_00018.spm

### TopoStats Version

2.1.2

### Python Version

3.11

### Operating System

MacOS Intel (pre-2021)

### Python Packages

asttokens==2.4.1
cheap-repr==0.5.1
contextlib2==21.6.0
contourpy==1.2.0
cycler==0.12.1
decorator==5.1.1
exceptiongroup==1.2.0
executing==2.0.1
fonttools==4.47.2
h5py==3.10.0
igor2==0.5.3
imageio==2.33.1
ipython==8.20.0
jedi==0.19.1
joblib==1.3.2
kiwisolver==1.4.5
lazy_loader==0.3
matplotlib==3.8.2
matplotlib-inline==0.1.6
networkx==3.2.1
numpy==1.26.3
packaging==23.2
pandas==2.2.0
parso==0.8.3
pexpect==4.9.0
pillow==10.2.0
prompt-toolkit==3.0.43
psutil==5.9.8
ptyprocess==0.7.0
pure-eval==0.2.2
pyfiglet==1.0.2
Pygments==2.17.2
pyparsing==3.1.1
pyspm==0.6.0
python-dateutil==2.8.2
pytz==2023.3.post1
PyYAML==6.0.1
ruamel.yaml==0.18.5
ruamel.yaml.clib==0.2.8
schema==0.7.5
scikit-image==0.22.0
scikit-learn==1.4.0
scipy==1.12.0
seaborn==0.13.1
six==1.16.0
snoop==0.4.3
stack-data==0.6.3
threadpoolctl==3.2.0
tifffile==2023.12.9
topofileformats==0.1.0
-e git+https://github.com/AFM-SPM/TopoStats.git@43410ab06bf366764271553a16f88d97f3c29999#egg=topostats
tqdm==4.66.1
traitlets==5.14.1
tzdata==2023.4
wcwidth==0.2.13

LHolmes2, 2024-02-08:
Hi @ns-rse , 
Thank you for the suggestion. I managed to do what you have suggested, however it doesn't seem to have fixed the issue. Still having strange traces, albeit different to the original traced file in this report. 
[image]

ns-rse, 2024-02-08:
Thanks for checking @LHolmes2 

I'm struggling to map the masks to where they should be but have an idea that this may be caused with how these are aligned with the overall image.

Its on the ToDo list.

MaxGamill-Sheffield, 2024-02-20:
I also think this might cause the "failed to skeletonise error" and dnatracing stats for these molecules to not be captured:
[image]

MaxGamill-Sheffield, 2024-11-13:
going to assume this has been fixed internally due to lack of response :)

closed 2024-11-13 by MaxGamill-Sheffield