AFM-SPM/TopoStats #805: [feature] : Remove smallest_grain_size_nm2 as a parameter from config (superfluous)

SylviaWhittle, 2024-02-20:
### Is your feature request related to a problem?

The `smallest_grain_size_nm2` parameter of the grains section of the config is superfluous, as `absolute_area_threshold` covers this and has understandably recently caused confusion within a group of users. 

It previously hadn't been removed since it was doing no harm, however we now have reason to action this given the aforementioned confusion.

### Describe the solution you would like.

Remove `smallest_grain_size_nm2` and allow the area thresholding step to take on the responsibility of removing very small objects from images, as it would already do if the noise wasn't already removed by the remove noise function.

### Describe the alternatives you have considered.

Keeping it in would likely cause continued confusion in the user-base.

ns-rse, 2024-03-05:
Remove from the following files...

+ `topostats/default_dict.yaml`
+ `topostats/validation.py`
+ `docs/configuration.md`
+ `topostats/grains.py`

The main work required is in `topostats/grains.py` where `smallest_grain_size_nm2` is used in a few places...

## `Grains.remove_noise()`

Defined on line [151](https://github.com/AFM-SPM/TopoStats/blob/8798977820ddbf165c258a27ac5ba533f6decbad/topostats/grains.py#L151) this method uses the value directly as an argument to `skimage.morphology.remove_small_objects()`.

The method is only ever defined and is not used anywhere so can be safely removed.

## Lower threshold for area thresholding.

Line [324](https://github.com/AFM-SPM/TopoStats/blob/8798977820ddbf165c258a27ac5ba533f6decbad/topostats/grains.py#L324)

```python
            self.directions[direction]["removed_noise"] = self.area_thresholding(
                self.directions[direction]["tidied_border"],
                [self.smallest_grain_size_nm2, None],
            )
```

I propose replacing the lower area threshold (`self.smallest_grain_size_nm2`) with `self.minimum_grain_size / (self.pixel_to_nm_scaling) ** 2`.

ns-rse, 2025-11-11:
Removed in 0c602ea1e8894c60f1e3c502037e13262a9570ea which was merged via #1123