AFM-SPM/TopoStats #847: Bending Angle Calculations

ns-rse, 2024-05-20:
Incorporate features from [JeanDu/BendingAngle-Upgraded](https://github.com/AFM-SPM/TopoStats/tree/JeanDu/BendingAngle-Upgraded) along with work under #168 into sub-modules under `topostats.measure`.

Jean-Du, 2024-05-20:
Main things to consider:

1. The relevant code is found in `measure_bending_angle` in dnatracing.py
2. The measurement is only performed once per molecule, so the results can be easily intergrated into the existing data output structure 
3. The site of measurement can be decided in two ways: the midpoint of the molecule, or the point of highest curvature within the central portion of the molecule; in the former case, **the bending angle can be measured independently from the curvature, so the only pre-requisite is successful tracing**
4. User input will be needed for parameters such as the length of the curve segement used on each side of the bending site; this can be added to the config file
5. It would also be worth adding the option to choose between the 'inside' and 'outside' angle of a bend (they should add up to 180°)

ns-rse, 2024-05-20:
+ Code will need refactoring and working into up-to-date framework which works with single grains rather than every function having a loop to iterate over the grains within an image.
+ Calculations should be configurable via `topostats/default_config.yaml` (the basis for generating input files)...
  + whether to make the calculations and whether to calculate `inside` or `outside`.
  + length of segment to be used either side of the curved segment.
  + choice of method for calculating bending angle (midpoint of molecule v point of highest curvature).
+ Results should be incorporated into HDF5 file output (is any additional output e.g. column in CSVs required?).