AFM-SPM/TopoStats #853: [Bug]: asd available channels not logged

MaxGamill-Sheffield, 2024-06-06:
### Checklist

- [X] Re-run analysis with `topostats process --core 1`.
- [X] Describe the bug.
- [X] Include the configuration file.
- [X] Copy of the output.
- [X] The exact command that failed. This is what you typed at the command line, including any options.
- [X] TopoStats version, this is reported by `topostats --version`
- [X] Operating System and Python Version

### Describe the bug

With the switch to` AFMReader` for asd i/o. The broad exception statement printing the warning for "Channel <channel> not found, skipping the image" is super-seeding the available channel message in `AFMReader.asd`

### Copy of the output

[image]

### Include the configuration file

N/A

### TopoStats Version

Git main branch

### Python Version

3.11

### Operating System

MacOS M1/M2 (post-2021)

ns-rse, 2024-06-06:
This hasn't been merged to `main` yet.

You can get information as to the version in the current Virtual Environment. Ideally a fresh `pip install -e .` should be done to ensure this is accurate.

```bash
❱ pip install -e .
❱ pip show topostats
Name: topostats
Version: 2.2.post1.dev190+g389fb1045
Summary: Automated Analysis for Atomic Force Microscopy Images of Biomolecules
Home-page: 
Author: 
Author-email: TopoStats Team <topostats@sheffield.ac.uk>, Alice Pyne <a.l.pyne@sheffield.ac.uk>, Sylvia Whittle <sylvia.whittle@sheffield.ac.uk>, Neil Shephard <n.shephard@sheffield.ac.uk>, Max Gamill <mcgamill1@sheffield.ac.uk>, Jean Du <mdu12@sheffield.ac.uk>
License: GNU Lesser GPLv3 only
Location: /home/neil/.virtualenvs/tmp-1b71451da46dd36/lib/python3.12/site-packages
Editable project location: /home/neil/work/git/hub/AFM-SPM/TopoStats
Requires: AFMReader, h5py, igor2, matplotlib, numpy, pandas, pyfiglet, pySPM, pyyaml, ruamel.yaml, schema, scikit-image, scipy, seaborn, snoop, tifffile, tqdm
Required-by: 
```

...and because we are dynamically setting version using `setuptools_scm` it appends the last tag with the `post1.dev190+g389fb1045` which indicates the number of commits ahead and the hash of the last commit on that branch.

Or as per the issue template `topostats --version` reports this and it can be copy pasted.

```
❱ topostats --version
Installed version of TopoStats: 2.2.post1.dev190+g389fb1045
```

MaxGamill-Sheffield, 2024-06-06:
Yep, just made a new PR for it. I forgot the keywords would close the accidental PR. Also because of testing on one branch `ns-rse/851...` then applying the changes to another `main` I thought the package list and commit version may have been more confusing.

PR #855 (2024-06-06): add orig err msg in broad channel try/except
Closes #853 

Adds 1 line reciting the original error message passed to a broad try/except.

For `asd` files, there available channels logging output from `AFMReader` is no longer super-seeded and output. Because the other filetypes don't use `AFMReader` yet, they have duplicate "available channel" outputs.
closed 2024-06-06 by MaxGamill-Sheffield