AFM-SPM/TopoStats #860: [feature] : Circularity Metric

MaxGamill-Sheffield, 2024-06-12:
### Is your feature request related to a problem?

Aim is to have a measure of circular-ness or roundness output in the grainstats.

### Describe the solution you would like.

Can be done by measuring the grains outer hull and using a ratio of its area to its perimeter (or circumference):

Roundness = 4π × Area / Perimeter^2

Will give a normalised value of 0 for non-circles and 1 for a perfect circle.

SylviaWhittle, 2024-06-12:
Note: A couple other similar metrics that have been used in the past are:

ratio of feret diameters - since a circle has a feret ratio of 1 and a line has a feret ratio of *big*. But this metric can be gamed.

Variance in distances of edge points from centroid

Though the geometrically derived metric above is likely the best in most cases

SylviaWhittle, 2025-02-10:
We would need the contained area of the grain - ie the middle of the donut plus the ring of the donut for the area that we use for the `4*π Area / Perimeter^2`