AFM-SPM/TopoStats #865: [Bug]: Matplotlib 3.9.0 register_cmap deprecation

SylviaWhittle, 2024-06-26:
### Checklist

- [X] Re-run analysis with `topostats process --core 1`.
- [X] Describe the bug.
- [X] Include the configuration file.
- [X] Copy of the output.
- [X] The exact command that failed. This is what you typed at the command line, including any options.
- [X] TopoStats version, this is reported by `topostats --version`
- [X] Operating System and Python Version

### Describe the bug

In 3.9.0, Matplotlib deprecates `register_cmap`. What should we do about this? Prevent use of 3.9 onwards or force it? Supporting both will be messy.

An [alternative](https://matplotlib.org/3.9.0/api/cm_api.html#matplotlib.cm.ColormapRegistry.register) is provided, in `matplotlib.colormaps.register`.

[Matplotlib API changes](https://matplotlib.org/3.9.0/api/prev_api_changes/api_changes_3.9.0.html) for 3.9.0, search for `register_cmap` to see the listing

### Copy of the output

N/A

### Include the configuration file

N/A

### To Reproduce

Try to import TopoStats while Matplotlib 3.9.0 is installed.

### TopoStats Version

Git main branch

### Python Version

3.12

### Operating System

MacOS M1/M2 (post-2021)

### Python Packages

Matplotlib 3.9.0

ns-rse, 2024-06-27:
I think this is a duplicate of #845 which was fixed with #846

No problems seen here...

```
(topostats) ❱ ipython 
~~~~~~~~~~~~ AUTORELOAD ENABALED ~~~~~~~~~~~~
Python 3.12.4 (main, Jun  7 2024, 06:33:07) [GCC 14.1.1 20240522]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.24.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import matplotlib

In [2]: matplotlib.__version__
Out[2]: '3.9.0'

In [3]: import topostats

In [4]: topostats.__version__
Out[4]: '2.2.2.dev5+gf5b40f762'

```

SylviaWhittle, 2024-06-27:
Closing as was a duplicate

closed 2024-06-27 by SylviaWhittle