AFM-SPM/TopoStats #866: [Bug]: Igor2 cannot be imported using Python 3.12

SylviaWhittle, 2024-06-26:
### Checklist

- [X] Re-run analysis with `topostats process --core 1`.
- [X] Describe the bug.
- [X] Include the configuration file.
- [X] Copy of the output.
- [X] The exact command that failed. This is what you typed at the command line, including any options.
- [X] TopoStats version, this is reported by `topostats --version`
- [X] Operating System and Python Version

### Describe the bug

When developing on python 3.12, Igor2 fails to import due to some issue in Igor2's code relating to passing arguments to `struct`.

I would make an issue on their repo but I cannot find one

### Copy of the output

Here is me using python 3.11 successfully, then python 3.12 unsuccessfully.

```
❯ pip show igor2
Name: igor2
Version: 0.5.8
Summary: interface for reading binary IGOR files
Home-page: 
Author: Paul Kienzle, W. Trevor King, Aurelien Jaquier, Zbigniew Jędrzejewski-Szmek, Paul Müller
Author-email: 
License: GNU Lesser General Public License v3 or later (LGPLv3+)
Location: /opt/homebrew/Caskroom/miniforge/base/envs/topo-igor-test-311/lib/python3.11/site-packages
Requires: numpy
Required-by: AFMReader, topostats
❯ python
Python 3.11.0 | packaged by conda-forge | (main, Jan 14 2023, 12:26:40) [Clang 14.0.6 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import igor2
>>> exit()
❯ conda deactivate
❯ conda activate 800_better_tracing
❯ pip show igor2
Name: igor2
Version: 0.5.5
Summary: interface for reading binary IGOR files
Home-page: 
Author: Paul Kienzle, W. Trevor King, Aurelien Jaquier, Zbigniew Jędrzejewski-Szmek, Paul Müller
Author-email: 
License: GNU Lesser General Public License v3 or later (LGPLv3+)
Location: /opt/homebrew/Caskroom/miniforge/base/envs/800_better_tracing/lib/python3.12/site-packages
Requires: numpy
Required-by: AFMReader, topostats
❯ pip install "igor2==0.5.8"
Collecting igor2==0.5.8
  Using cached igor2-0.5.8-py3-none-any.whl.metadata (2.7 kB)
Requirement already satisfied: numpy>=1.19.2 in /opt/homebrew/Caskroom/miniforge/base/envs/800_better_tracing/lib/python3.12/site-packages (from igor2==0.5.8) (1.26.4)
Using cached igor2-0.5.8-py3-none-any.whl (32 kB)
Installing collected packages: igor2
  Attempting uninstall: igor2
    Found existing installation: igor2 0.5.5
    Uninstalling igor2-0.5.5:
      Successfully uninstalled igor2-0.5.5
Successfully installed igor2-0.5.8
❯ python
Python 3.12.0 | packaged by conda-forge | (main, Oct  3 2023, 08:36:57) [Clang 15.0.7 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import igor2
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/homebrew/Caskroom/miniforge/base/envs/800_better_tracing/lib/python3.12/site-packages/igor2/__init__.py", line 10, in <module>
    from . import binarywave
  File "/opt/homebrew/Caskroom/miniforge/base/envs/800_better_tracing/lib/python3.12/site-packages/igor2/binarywave.py", line 114, in <module>
    BinHeader1 = _Structure(  # `version` field pulled out into Wave
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: Struct() takes at most 1 keyword argument (2 given)
>>> exit()
     ~/Documents/TopoStats     maxgamill-sh…tter-tracing *43 !4 ?2                                                                                          7s    4.05G    0B    800_better_tracing    23:31:41   
❯ 
```

### Include the configuration file

N/A

### To Reproduce

Import Igor2 in python 3.12

### TopoStats Version

Git main branch

### Python Version

3.12

### Operating System

MacOS M1/M2 (post-2021)

### Python Packages

Igor2 in versions 0.5.8 (latest) and 0.5.5 (just to check if it's only recent, as latest update was June 17th)

ns-rse, 2024-06-27:
See [#15 Check Python 3.12 version compatibility](https://github.com/AFM-analysis/igor2/issues/15) which is suggested as a duplicate of my original report of Python 3.12 compatability ([#9 Tests fail under Python 3.12](https://github.com/AFM-analysis/igor2/issues/9)).

Looking at your output you are using `python-3.12.0` and the fix to the underlying `cpython` was made in `python-3.12.2`.

My environment where I don't see this problem is using `python-3.12.4` and I can import both `igor2-0.5.6` and `igor2-0.5.8` (having just upgraded to check).

## Solution

Probably upgrade to `python>=3.12.2`.

ns-rse, 2024-06-27:
Closing, not TopoStats issue and solution identified.

closed 2024-06-27 by ns-rse