AFM-SPM/TopoStats #887: [feature] : message in terminal that says which image causes an error when processing multiple files at the same time

TFirth2, 2024-08-15:
### Is your feature request related to a problem?

Currently, to find an image or images that cause an error in TopoStats you have to run 1 core which stops on the image that causes the error and this is inefficient. If there are multiple error causing images that can be very time consuming.

### Describe the solution you would like.

 It would be good to run the entire folder and then get a message saying that X image or X image caused the error so these can be dealt with more effectively.

### Describe the alternatives you have considered.

I currently batch process image with grain stats off to just view the images and manually remove images that are cut off, blanks or otherwise not good. I then re-run TopoStats for grain stats but then often get stuck on images that won't process to give grain stats.

ns-rse, 2024-09-16:
Thanks for highlighting this @TFirth2 

I can fully appreciate how much of a pain this is.

Its like this because of [technical debt](https://en.wikipedia.org/wiki/Technical_debt) where we were rushing to get something working quickly and implemented a broad and crude solution and haven't gone back and fixed it yet.

When we write code sometimes errors arise, but we want to handle those errors. In Python that can be done with the `try: ... except: ...` construct where some code is tried and if an exception (also known as an error) arises it does what is defined under the `except:` clause.

We have a broad `try: ... except: ...` around grain finding (see lines [166-184](https://github.com/AFM-SPM/TopoStats/blob/70d02bd811285efcb6c72bdef854b148f414b951/topostats/processing.py#L161) if interested) although I notice its currently commented out on the `main` branch which is possibly a hangover from someone debugging (currently checking with @SylviaWhittle ).

The challenge is knowing/anticipating what the possible errors that can occur might be and with only `minicircle.spm` to work with on developing tests there wasn't much scope for knowing what might crop up.

If you, or anyone else in the lab, have some sample images which cause errors it would be really useful for the developers to have them so we can see what they cause and write finer-grained code to capture these and, hopefully, allow processing of other images to continue.

ns-rse, 2024-11-19:
Has the reduction in logging information addressed this I wonder?

Are there batches of files we could test with please?

Feels like it might be possible to close this out for the release.

ns-rse, 2024-11-27:
@TFirth2 We think this is now handled as desired as we have reduced the amount of logging information output and report warnings when specific images fail.

Are you able to test against the current `main` branch on a set of files with at least one (ideally a few) that are known to fail and a few that are known to pass please?

If so please report back here on whether the failing images are correctly identified and paste any and all output so we can see what we're missing.

Thanks in advance,

@ns-rse

ns-rse, 2024-12-03:
**bump**

Have you had an opportunity to check that the current `main` branch gives sufficient feedback yet @TFirth2 ?

ns-rse, 2024-12-11:
@MaxGamill-Sheffield to chase up @TFirth2 to see if current state of messages covers the requested behaviour.

MaxGamill-Sheffield, 2024-12-11:
@TFirth2 has agreed that the changes to how exceptions were handled in the tidy-up post tracing additions are a suitable solution and this can be closed.

However, TopoStats may still halt as we may have missed some areas to use the broard try-except statements so I will update the `bug_report.yaml` to say: "If TopoStats halts processing, re-run analysis with `topostats process --core 1` to find the offending file."

closed 2024-12-12 by MaxGamill-Sheffield