AFM-SPM/TopoStats #935: [Bug]: wrong magnitude

AdriSP2, 2024-10-09:
### Checklist

- [X] Re-run analysis with `topostats process --core 1`.
- [X] Describe the bug.
- [X] Include the configuration file.
- [X] Copy of the output.
- [X] The exact command that failed. This is what you typed at the command line, including any options.
- [X] TopoStats version, this is reported by `topostats --version`
- [X] Operating System and Python Version

### Describe the bug

When processing some Highspeed images, for some reason the magnitude of the height is about 10 times higher than the real one.
For instance, while manually analyzing some images you get a height of around 1.5-1.8nm the output is giving around 11-8nm being 8 if I increase the above threshold for grains as maximum as possible for the mask to fit.
Could it be somehow that the channel is being read wrong? Using another set of images it worked fine

### Copy of the output

[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [grains.py] [318] [230523-01-0048_nigericina_61-1017_1015] : Coloured regions
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [grains.py] [416] [230523-01-0048_nigericina_61-1017_1015] : Extracted bounding boxes (above)
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [171] [230523-01-0048_nigericina_61-1017_1015] : Grains found for direction above : 2
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [186] [230523-01-0048_nigericina_61-1017_1015] : Plotting Grain Finding Images
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [188] [230523-01-0048_nigericina_61-1017_1015] : Plotting above Grain Finding Images
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [194] [230523-01-0048_nigericina_61-1017_1015] : Plotting mask_grains image
[Wed, 02 Oct 2024 18:08:55] [WARNING ] [topostats] [io.py] [1110] Item 'grain_trace_data' is None. Skipping.
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [io.py] [1138] Cannot save key 'grain_trace_data' to HDF5. Item type: <class 'NoneType'>. Skipping. One of data, shape or dtype must be specified
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [194] [230523-01-0048_nigericina_61-1017_1015] : Plotting labelled_regions_01 image
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [194] [230523-01-0048_nigericina_61-1017_1015] : Plotting tidied_border image
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [194] [230523-01-0048_nigericina_61-1017_1015] : Plotting removed_noise image
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [194] [230523-01-0048_nigericina_61-1017_1015] : Plotting removed_small_objects image
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [194] [230523-01-0048_nigericina_61-1017_1015] : Plotting labelled_regions_02 image
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [194] [230523-01-0048_nigericina_61-1017_1015] : Plotting coloured_regions image
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [plottingfuncs.py] [307] [230523-01-0048_nigericina_61-1017_1013_above_masked] : Image saved to : output\processed\230523-01-0048_nigericina_61-1017_1013_above_masked.png | DPI: 100
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [280] [230523-01-0048_nigericina_61-1017_1013] : *** Grain Statistics ***
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [grainstats.py] [220] [230523-01-0048_nigericina_61-1017_1013] : Processing grain: 0
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [plottingfuncs.py] [307] [230523-01-0048_nigericina_61-1017_1014_above_masked] : Image saved to : output\processed\230523-01-0048_nigericina_61-1017_1014_above_masked.png | DPI: 100
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [280] [230523-01-0048_nigericina_61-1017_1014] : *** Grain Statistics ***
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [grainstats.py] [220] [230523-01-0048_nigericina_61-1017_1014] : Processing grain: 0
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [plottingfuncs.py] [307] [230523-01-0048_nigericina_61-1017_1016] : Image saved to : output\processed\230523-01-0048_nigericina_61-1017_1016.png | DPI: 100
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [162] [230523-01-0048_nigericina_61-1017_1016] : *** Grain Finding ***
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [grains.py] [358] [230523-01-0048_nigericina_61-1017_1016] : Thresholding method (grains) : absolute
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [grains.py] [367] [230523-01-0048_nigericina_61-1017_1016] : Finding above grains, threshold: (5)
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [utils.py] [165] [230523-01-0048_nigericina_61-1017_1016] : Masking (above) Threshold: 5
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [grains.py] [169] [230523-01-0048_nigericina_61-1017_1016] : Labelling Regions
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [grains.py] [147] [230523-01-0048_nigericina_61-1017_1016] : Tidying borders
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [grains.py] [386] [230523-01-0048_nigericina_61-1017_1016] : Removing noise (above)
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [grains.py] [287] [230523-01-0048_nigericina_61-1017_1016] : Area thresholding grains | Thresholds: L: 1.12,U: 5625.00 px^2, L: 0.50, U: 2500.00 nm^2.
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [grains.py] [392] [230523-01-0048_nigericina_61-1017_1016] : Removing small / large grains (above)
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [grains.py] [287] [230523-01-0048_nigericina_61-1017_1016] : Area thresholding grains | Thresholds: L: 2.25,U: 6750.00 px^2, L: 1.00, U: 3000.00 nm^2.
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [grains.py] [169] [230523-01-0048_nigericina_61-1017_1016] : Labelling Regions
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [grains.py] [411] [230523-01-0048_nigericina_61-1017_1016] : Region properties calculated (above)
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [grains.py] [318] [230523-01-0048_nigericina_61-1017_1016] : Coloured regions
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [grains.py] [416] [230523-01-0048_nigericina_61-1017_1016] : Extracted bounding boxes (above)
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [171] [230523-01-0048_nigericina_61-1017_1016] : Grains found for direction above : 1
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [186] [230523-01-0048_nigericina_61-1017_1016] : Plotting Grain Finding Images
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [188] [230523-01-0048_nigericina_61-1017_1016] : Plotting above Grain Finding Images
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [194] [230523-01-0048_nigericina_61-1017_1016] : Plotting mask_grains image
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [194] [230523-01-0048_nigericina_61-1017_1016] : Plotting labelled_regions_01 image
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [194] [230523-01-0048_nigericina_61-1017_1016] : Plotting tidied_border image
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [194] [230523-01-0048_nigericina_61-1017_1016] : Plotting removed_noise image
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [194] [230523-01-0048_nigericina_61-1017_1016] : Plotting removed_small_objects image
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [194] [230523-01-0048_nigericina_61-1017_1016] : Plotting labelled_regions_02 image
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [194] [230523-01-0048_nigericina_61-1017_1016] : Plotting coloured_regions image
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [grainstats.py] [311] [230523-01-0048_nigericina_61-1017_1013] : Height profiles extracted.
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [480] [230523-01-0048_nigericina_61-1017_1013] Calculation of DNA Tracing disabled, returning grainstats data frame.
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [processing.py] [647] [230523-01-0048_nigericina_61-1017_1013] : *** Image Statistics ***
[Wed, 02 Oct 2024 18:08:55] [INFO    ] [topostats] [io.py] [1189] [230523-01-0048_nigericina_61-1017_1013] : Saving image to .topostats file
[Wed, 02 Oct 2024 18:08:55] [WARNING ] [topostats] [io.py] [1110] Item 'grain_trace_data' is None. Skipping.
[... cut ...]
seaborn==0.13.2
six==1.16.0
snoop==0.4.3
stack-data==0.6.3
threadpoolctl==3.5.0
tifffile==2024.7.2
topostats==2.2.1
tqdm @ file:///C:/b/abs_f76j9hg7pv/croot/tqdm_1679561871187/work
traitlets==5.14.3
truststore @ file:///C:/b/abs_55z7b3r045/croot/truststore_1695245455435/work
typing_extensions==4.12.2
tzdata==2024.1
urllib3 @ file:///C:/b/abs_9cmlsrm3ys/croot/urllib3_1698257595508/work
wcwidth==0.2.13
win-inet-pton @ file:///C:/ci_310/win_inet_pton_1642658466512/work
win32-setctime==1.1.0
zstandard==0.19.0

ns-rse, 2024-10-09:
Thanks for the :lady_beetle: report @AdriSP2 

The first thing to check is whether this problem persists in the most recent version of TopoStats which is [v2.2.1](https://github.com/AFM-SPM/TopoStats/releases/tag/v2.2.1).

Ideally this should be done in a clean virtual environment with the latest versions of all packages installed. This can be done after creating and activating your virtual environment using...

```bash
pip install --no-cache-dir topostats==2.2.1
```

I notice also that you've indicated that you are running GNU/Linux but some of the package paths that are listed by `pip freeze` point to files that look like they are on a Windows system as they include `C:/` are you running Windows Subsystem for Linux perhaps?

AdriSP2, 2024-10-09:
Hi, I got an issue opening the tiket and it looks like the dropdown menues went to the default position
Topostats version 2.2.1
Python 3.10.13
OS: Windows

let me see if i can edit it from the top so it does not create confusion

AdriSP2, 2024-10-24:
The issue has been fixed since using version 2.2.2.dev938+g5d01d437.d20241024.
Probably this issue was not caused by TopoStats but by trying to circumvent the issue #942.
In order to circumvent issue #942, before it was fixed, the working method was to crop the HS-AFM video and eliminate the frames that were having no object to be detected.
It seems that in this process of cropping, the program used (UMEX Viewer) was changing somehow the data of height. In the program everything was alright but it seems TopoStats was detecting it somehow differently.
Anyway, using the original full videos made the issue disappear.
Thanks!

closed 2024-10-24 by AdriSP2